JvanKatwijk/qt-dab #10: Dropouts due to CPU meter?

andimik, 2017-02-12:
It seems never versions produce drop-out after inserting the CPU meter.

JvanKatwijk, 2017-02-12:
Can you be more specific? The load is only computed once every two seconds
best
j

andimik, 2017-02-13:
I've tested it again. Seems playback from an sdr file caused that strange dropout. I can reproduce them (older version no dropout, newer with)

You could try to load an sdr file, probably with the samples I sent you from Slovenia.

Very funny, there are absolutely no dropouts when I press the "audio" button, write a wav-file and listen to that.

JvanKatwijk, 2017-02-13:
I do not have dropouts,
best
j

zygmund2000, 2017-02-15:
I can it confirm too, sound dropouts occurs randomly qt-dab, old sdr-j works without dropouts.

JvanKatwijk, 2017-02-15:
What is the cp load when these dropouts happen. In my case the cpuload is
app 20% and I do not notice dropouts
J

zygmund2000, 2017-02-15:
CPU load is low all time, this droputs are not related to CPU load, looks like audio speed don't match to soundcard speed or something like this, because sound don't stop but change time.

JvanKatwijk, 2017-02-16:
You had a remark on drop outs of sound.
There are a few possibilities that can cause drop outs in sound.
a. if the DAB+ frames have crc errors than, obviously, the containing data
is discarded and the frames - for a DAB+ frame 120 msec -
do not contribute to the output.
b. if the audio units in the DAB frames contain errors, then the
contribution of such an audio unit (24 msec) is discarded
c. if the clock on the soundcard differs from the clock in the input
device, then it might happen that soundframes are either "early"
or "late" arriving at the sound engine.
d. the software driving the soundcard is slow and needs a particular
buffersize for a continuous soundstream.

a) and b) happen, but that shows on the markers on the technical data
widget.
c. might happen, but unless the difference is large, the effect sould be
limited
d. happens from time to time. I instrumented my code with a counter telling
how many samples were missing with different buffersizes
   On my RPI this happens when the buffersize is too small,
   On my laptop this happens - with a dropout of 24 msec on say every 15
seconds - with too small or too large a buffersize

The parameter here is the setting of the latency, a parameter that can be
set in the .ini file. On my laptop a value 1 or 2 gives as result
no noticeable dropout - provided they are not caused by earlier mentioned
reasons - when running files or input over minutes
Setting the latency parameter to 0, i.e. a buffersize indicated as 0, such
that the underlying sound engine will define its own size,
gives dropouts, app 24 msec in about 15 seconds.

My suggestion is to experiment a little with the latency parameter, and if
needed I can create a version with a smaller
increment in buffersize for higher latency values (one issue on my laptop
is that the default setting for all kinds of Latency
in the portaudio module that I am using is 0 in all cases, so I "invented"
my own buffersize).

best
jan

zygmund2000, 2017-02-16:
Just add line to qt-dab.ini latency=1?
    
    On 16.02.2017 13:02, JvanKatwijk wrote:
    
    You had a remark on drop outs of sound.
      There are a few possibilities that can cause drop outs in sound.
      a. if the DAB+ frames have crc errors than, obviously, the
      containing data
      is discarded and the frames - for a DAB+ frame 120 msec -
      do not contribute to the output.
      b. if the audio units in the DAB frames contain errors, then the
      contribution of such an audio unit (24 msec) is discarded
      c. if the clock on the soundcard differs from the clock in the
      input
      device, then it might happen that soundframes are either "early"
      or "late" arriving at the sound engine.
      d. the software driving the soundcard is slow and needs a
      particular
      buffersize for a continuous soundstream.
      
      a) and b) happen, but that shows on the markers on the technical
      data
      widget.
      c. might happen, but unless the difference is large, the effect
      sould be
      limited
      d. happens from time to time. I instrumented my code with a
      counter telling
      how many samples were missing with different buffersizes
      On my RPI this happens when the buffersize is too small,
      On my laptop this happens - with a dropout of 24 msec on say every
      15
      seconds - with too small or too large a buffersize
      
      The parameter here is the setting of the latency, a parameter that
      can be
      set in the .ini file. On my laptop a value 1 or 2 gives as result
      no noticeable dropout - provided they are not caused by earlier
      mentioned
      reasons - when running files or input over minutes
      Setting the latency parameter to 0, i.e. a buffersize indicated as
      0, such
      that the underlying sound engine will define its own size,
      gives dropouts, app 24 msec in about 15 seconds.
      
      My suggestion is to experiment a little with the latency
      parameter, and if
      needed I can create a version with a smaller
      increment in buffersize for higher latency values (one issue on my
      laptop
      is that the default setting for all kinds of Latency
      in the portaudio module that I am using is 0 in all cases, so I
      "invented"
      my own buffersize).
      
      best
      jan

andimik, 2017-02-16:
Yes, with this parameter the dropouts are more, less or even gone (for some seconds), depending on the entry. I'm alway replaying the same sdr-file, so I know when the dropouts appear.

When dropouts occur, there is sometimes this message in the command window:

`ALSA lib pcm.c:7963:(snd_pcm_recover) underrun occurred
`

The SDR-J had 

`Suggested size for outputbuffer = 5120`

qt-dab has

`Suggested size for outputbuffer = 15360
`

zygmund2000, 2017-02-17:
Suggested size for outputbuffer =
          5120 is better, maybe skpis are too short to hear, I have
          hardware calibrated rtlsdr and data speed shoud be very exact,
          soundcard is also quite exact, dab signal is very very exact
          at the moment. No console errors.
      
    
    On 16.02.2017 23:48, andimik wrote:
    
    
      Yes, with this parameter the dropouts are more, less or even
        gone (for some seconds), depending on the entry. I'm alway
        replaying the same sdr-file, so I know when the dropouts appear.
      When dropouts occur, there is sometimes this message in the
        command window:
      ALSA lib pcm.c:7963:(snd_pcm_recover) underrun occurred
      The SDR-J had
      Suggested size for outputbuffer = 5120
      qt-dab has
      Suggested size for outputbuffer = 15360
      —
          it on GitHub, or mute
          the thread.

JvanKatwijk, 2017-02-17:
Two changes/additions
a. the granularity for the pcm buffersize is made smaller, now the size =
latency * 5 * 256, default value for latency parameter is 1.
b. a "secret" command line frag is added, -T. When selected, there will be
a continuous report on the amount of frameerrors and the amount of missed
samples.
hope this helps
 jan

zygmund2000, 2017-02-17:
Suggested size for outputbuffer = 1280

we have now DAB+
 frameErrors = 0, missed samples for audio 44800
 frameErrors = 0, missed samples for audio 640
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 640
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0

JvanKatwijk, 2017-02-17:
and in the longer run?

JvanKatwijk, 2017-02-17:
I'll make the size of the granules 128.
j

2017-02-17 11:37 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:

> and in the longer run?
>
>
>
>> Suggested size for outputbuffer = 1280
>>
>> we have now DAB+
>> frameErrors = 0, missed samples for audio 44800
>> frameErrors = 0, missed samples for audio 640
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 640
>> frameErrors = 0, missed samples for audio 0
>> frameErrors = 0, missed samples for audio 0
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

JvanKatwijk, 2017-02-17:
... and you can always try "latency=0", then the portaudio layer looks for
a default size
j

2017-02-17 11:38 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:

> I'll make the size of the granules 128.
> j
>
>
> 2017-02-17 11:37 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:
>
>> and in the longer run?
>>
>>
>>
>>> Suggested size for outputbuffer = 1280
>>>
>>> we have now DAB+
>>> frameErrors = 0, missed samples for audio 44800
>>> frameErrors = 0, missed samples for audio 640
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 640
>>> frameErrors = 0, missed samples for audio 0
>>> frameErrors = 0, missed samples for audio 0
>>>
>>>
>>
>>
>>
>> Jan van Katwijk
>>
>>
>> +31 (0)15 3698980 <+31%2015%20369%208980>
>> +31 (0) 628260355 <+31%206%2028260355>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

zygmund2000, 2017-02-17:
still no errors, I have pulseaudio 10 and 48khz default sample rate, now in non realtime mode for tests.

andimik, 2017-02-18:
Now the dropouts disappeared. Thanks!

HawkingJan, 2017-03-20:
On my 'Rasberry Pi 2 Model B I also have problems with hearable dropouts

```
 frameErrors = 4, missed samples for audio 215040
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 76416
 frameErrors = 0, missed samples for audio 73728
 frameErrors = 0, missed samples for audio 55296
 frameErrors = 0, missed samples for audio 1341440
 frameErrors = 0, missed samples for audio 361344
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 709824
 frameErrors = 0, missed samples for audio 476800
 frameErrors = 0, missed samples for audio 229376
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 5248
 frameErrors = 0, missed samples for audio 79872
 frameErrors = 0, missed samples for audio 87040
 frameErrors = 0, missed samples for audio 968704

```

I tried latency=0, latency=2, latency=4 and latency=8.

In the "technical data" window a CPU usage between 45 and 55 (%?)  is shown and the four bars at the bottom are all at 100%. Does anybody have some ideas how to solve this problem?

andimik, 2017-03-30:
It seems the dropouts disappeared with newer versions. But I will check again.

HawkingJan, 2017-05-21:
I compiled the latest version (0.999) and still have mayn of dropouts :-/ (using a dabstick on my Raspberry Pi)

```
 frameErrors = 3, missed samples for audio 0
 frameErrors = 0, missed samples for audio 0
 frameErrors = 0, missed samples for audio 5151
 frameErrors = 5, missed samples for audio 390
 frameErrors = 4, missed samples for audio 5286
 frameErrors = 1830, missed samples for audio 6387344
 frameErrors = 75, missed samples for audio 95910
```

@JvanKatwijk can you comment on the meaning of these numbers and on the indicator "MOT decoding" (shown in the technical details)?

JvanKatwijk, 2017-05-21:
Do you have dropouts when you comment out things like spectrum,
technicaldata etc etc

wrt the numbers:
they mean what is sais, frameerrors is the number of detected errors in the
frames and missed samples is missed samples in the
stream of 48000 samples/second
The numbers are not to be taken too seriously, the caounters are not always
reset. The numbers are merely for me to aid me
in some special checking .

HawkingJan, 2017-05-21:
Thanks for your quick reply.
I only used "dabstick" as a device in the qt-dab.pro. I now additionally commented out "spectrum", " TECHNICAL_DATA", " MOT_DATA" (not sure what it is about?) and "MSC_DATA__" (it said that one does not want it, why is it then not commented out by default?) and "__QUALITY" (just to be sure) and recompiled. Still lots of dropouts :-/

HawkingJan, 2017-05-23:
I now compiled the latest version and removed "spectrum" (but kept the other entries listed in my previous post, maybe you can comment on MOT_DATA and MSC_DATA_)?
I'm running my Pi as a headless device over ssh. I assume that the CPU usage value shown in the technical details is in percent? Then I'm using between 40-52%. Which latency value do you use/recommend?

I'm still not sure about the numbers printed using the -T option. Is the following interpretation correct?

frameErrors = 0, missed samples for audio 0 <-- everything OK
frameErrors = 5, missed samples for audio 0 <-- there were errors in the frame that could be corrected, audio signal should be OK.
frameErrors = 20, missed samples for audio 55564 <--- there were errors in the frame that could be corrected but audio signal still has issues (why?)
frameErrors = 0, missed samples for audio 5757 <-- ???
CRC failure with dab+ frame 2 (3) <-- there were errors in the frame that could not be corrected (this error happens rarely)

JvanKatwijk, 2017-05-24:
Interpretation is correct. There are three tests on DAB+ frames. The DAB
frames themselves are built up from 5 subsequent segments,
the first test is whether there is a frame or not, then with an RS
algorithm there is error correction to up to 5 byte errors in a 120 byte
frame,
If there are more than 5 errors the frame is rejected.
The implementation of RS needs revision since occasionally it happens that,
while RS states that all errors were corrected, the subsequent CRC fails
(which obviously should not happen).

A DAB+ frame contains 120msec of encoded sound, so when a frame is missing,
you miss for 120 msec samples, which with a samplerate of 48000 is about
5500

The counting of the amount of errors is not correctly done, and the exact
values themselves are - at least for me - not very interesting.

A timer polls the values, since there is a delay in the output (output is
sent to a buffer and asynchrously read) it might happen that frame errors
!= 0 and missed samples == 0

There is no display of the amount of corrected errors.

JvanKatwijk, 2017-05-24:
Wrt storing the coarse offset: having run a number of tests with different
devices, it shows that the offset is not constant over the different
channels, i.e. for the channels we are receiving here, the sdrplay that I
run shows an offset -1 for channel 5B and 2 for channel 12C.
For the dabstick is varies between -11 for channel 11C and -12 for channel
12C

Setting the ppm offset for the dabstick such that the coarse offset = 0 is
better, in that case (ppm offset = 84) I found a coarse
offset 0 on channels 8A, 11C and 12 C (those are the channels that can be
received with the dabstick, for 5B I need a different antenna
although I can receive that without problems with sdrplay and airspy)

2017-05-24 8:55 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Interpretation is correct. There are three tests on DAB+ frames. The DAB
> frames themselves are built up from 5 subsequent segments,
> the first test is whether there is a frame or not, then with an RS
> algorithm there is error correction to up to 5 byte errors in a 120 byte
> frame,
> If there are more than 5 errors the frame is rejected.
> The implementation of RS needs revision since occasionally it happens
> that, while RS states that all errors were corrected, the subsequent CRC
> fails (which obviously should not happen).
>
> A DAB+ frame contains 120msec of encoded sound, so when a frame is
> missing, you miss for 120 msec samples, which with a samplerate of 48000 is
> about 5500
>
> The counting of the amount of errors is not correctly done, and the exact
> values themselves are - at least for me - not very interesting.
>
> A timer polls the values, since there is a delay in the output (output is
> sent to a buffer and asynchrously read) it might happen that frame errors
> != 0 and missed samples == 0
>
> There is no display of the amount of corrected errors.
>
>
>
>> I now compiled the latest version and removed "spectrum" (but kept the
>> other entries listed in my previous post, maybe you can comment on MOT_DATA
>> and MSC_DATA_)?
>> I'm running my Pi as a headless device over ssh. I assume that the CPU
>> usage value shown in the technical details is in percent? Then I'm using
>> between 40-52%. Which latency value do you use/recommend?
>>
>> I'm still not sure about the numbers printed using the -T option. Is the
>> following interpretation correct?
>>
>> frameErrors = 0, missed samples for audio 0 <-- everything OK
>> frameErrors = 5, missed samples for audio 0 <-- there were errors in the
>> frame that could be corrected, audio signal should be OK.
>> frameErrors = 20, missed samples for audio 55564 <--- there were errors
>> in the frame that could be corrected but audio signal still has issues
>> (why?)
>> frameErrors = 0, missed samples for audio 5757 <-- ???
>> CRC failure with dab+ frame 2 (3) <-- there were errors in the frame that
>> could not be corrected (this error happens rarely)
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

closed 2018-02-12 by JvanKatwijk