JvanKatwijk/qt-dab #8: qmake qtdab.pro error

WOlsen80, 2017-02-12:
Hi,

I'm hoping that some could help with the error below?

Thanks

pi@raspberrypi:~/qt-dab $ qmake qt-dab.pro
pi@raspberrypi:~/qt-dab $ make
g++ -c -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -DQT_WEBKIT -DTECHNICAL_DATA -DMSC_DATA__ -DTRY_EPG -DHAVE_SPECTRUM -DHAVE_DABSTICK -DHAVE_RTL_TCP -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/epg -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Isrc/input -Isrc/input/rawfiles -Isrc/input/wavfiles -I/usr/local/include -I/usr/include/qt5/qwt -Iincludes/scopes-qwt6 -Iincludes/scopes-qwt6 -Ioptional-scope -I/home/jan/rtl-sdr/include -Isrc/input/dabstick -Isrc/input/rtl_tcp -I. -I. -o radio.o radio.cpp
In file included from optional-scope/spectrum-handler.h:42:0,
                 from radio.cpp:71:
./ui_scopewidget.h:19:22: fatal error: qwt_plot.h: No such file or directory
 #include "qwt_plot.h"
                      ^
compilation terminated.
Makefile:591: recipe for target 'radio.o' failed
make: *** [radio.o] Error 1

JvanKatwijk, 2017-02-12:
You probably have the wrong path for qwt.
The easiest thing to do would be to comment "spectrum" out, then you would
not get qwt to be included
j

andimik, 2017-02-12:
In my case I've added the correct path to the qt-dab.pro-file:

`INCLUDEPATH	+= /usr/local/include   /home/andreas/qwt-6.1.3/src/`

Then it works with the spectrum ...

andimik, 2017-02-12:
Seems `INCLUDEPATH	+= /usr/local/include /usr/include/qwt` is working as well

JvanKatwijk, 2017-02-12:
The correct include path strongly depends on the distribution. I have see
/usr/include/qwt /usr/include/qt5/qwt /usr/include/qt4/qwt
so it is pretty hard to have a generci solution

j

andimik, 2017-02-12:
Sorry, I had several installations of qwt (before I found out why building of sdr-j failed).

Now I found the **correct** path, where it is installed, see also http://qwt.sourceforge.net/qwtinstall.html

`/usr/local/qwt-6.1.3`

See qwtconfig.pri in the qwt-directory.

`QWT_INSTALL_PREFIX    = /usr/local/qwt-$$QWT_VERSION`

WOlsen80, 2017-02-13:
I commented out specrum and everything ran perfectly and i'm not up and running! 

I'm putting together a car stereo replacement built around a RPi3 as a hobby project and Qt-DAB is just what I've been looking for to get DAB radio into my car.

Thanks for the help
W

JvanKatwijk, 2017-02-13:
Good to hear.

JvanKatwijk, 2017-02-13:
That does not solve the problem, in a mixed qt4/qt5 environment I find qwt
at /usr/include/qt5/qwt
j

andimik, 2017-02-13:
Yes, I agree with you, there are some possibilities. But why not list all these paths? I'm pretty sure questions like this above would be much fewer...

JvanKatwijk, 2017-02-13:
Suggestions?
best
j

andimik, 2017-02-13:
Very easy suggestion: Change the line 169 to 

`INCLUDEPATH	+= /usr/local/include /usr/include/qt4/qwt /usr/include/qt5/qwt /usr/include/qt4/qwt /usr/include/qwt`

It works even if one of the paths does not exist, for example some lines below:

`INCLUDEPATH	+= /home/jan/rtl-sdr/include`

;-)

andreygursky, 2017-03-14:
@andimik, is the double occurrence of ```/usr/include/qt4/qwt``` a typo?

andimik, 2017-03-14:
Well, I don't know at the moment, but I think you're right. But wrong paths seem to be ignored (see above)

closed 2018-02-12 by JvanKatwijk