JvanKatwijk/qt-dab #12: make fails to compile ofdm-decoder

steeley, 2017-02-13:
get this error when running make (Ubuntu 16)

src/ofdm/ofdm-decoder.cpp: In member function ‘void ofdmDecoder::decodeFICblock(int32_t)’:
src/ofdm/ofdm-decoder.cpp:253:1: warning: label ‘fftlabel’ defined but not used [-Wunused-label]
 fftlabel:
 ^~~~~~~~
src/ofdm/ofdm-decoder.cpp:263:1: warning: label ‘toBitsLabel’ defined but not used [-Wunused-label]
 toBitsLabel:
 ^~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp:289:1: warning: label ‘handlerLabel’ defined but not used [-Wunused-label]
 handlerLabel:
 ^~~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp: In member function ‘void ofdmDecoder::decodeMscblock(int32_t)’:
src/ofdm/ofdm-decoder.cpp:317:1: warning: label ‘fftLabel’ defined but not used [-Wunused-label]
 fftLabel:
 ^~~~~~~~
src/ofdm/ofdm-decoder.cpp:322:1: warning: label ‘toBitsLabel’ defined but not used [-Wunused-label]
 toBitsLabel:
 ^~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp:336:1: warning: label ‘handlerLabel’ defined but not used [-Wunused-label]
 handlerLabel:;
 ^~~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp: At global scope:
src/ofdm/ofdm-decoder.cpp:246:5: warning: ‘cnt’ defined but not used [-Wunused-variable]
 int cnt = 0;
     ^~~
Makefile:603: recipe for target 'ofdm-decoder.o' failed
make: *** [ofdm-decoder.o] Error 1

JvanKatwijk, 2017-02-13:
Thanks

to me it seems you get more output than this, these are only warnings on
labels that are there to enhance readability.
Send the full output pls
best
j

steeley, 2017-02-13:
Oops sorry.. here is the full output from make:

/usr/lib/x86_64-linux-gnu/qt5/bin/uic src/input/filereader-widget.ui -o ui_filereader-widget.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic forms/dabradio.ui -o ui_dabradio.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic forms/technical_data.ui -o ui_technical_data.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic src/input/dabstick/dabstick-widget.ui -o ui_dabstick-widget.h
g++ -c -m64 -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -fPIC -DTECHNICAL_DATA -DTRY_EPG -DHAVE_DABSTICK -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/epg -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Isrc/input -Isrc/input/rawfiles -Isrc/input/wavfiles -isystem /usr/local/include -I/home/jan/rtl-sdr/include -Isrc/input/dabstick -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtXml -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.cpp
g++ -c -m64 -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -fPIC -DTECHNICAL_DATA -DTRY_EPG -DHAVE_DABSTICK -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/epg -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Isrc/input -Isrc/input/rawfiles -Isrc/input/wavfiles -isystem /usr/local/include -I/home/jan/rtl-sdr/include -Isrc/input/dabstick -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtXml -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o radio.o radio.cpp
g++ -c -m64 -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -fPIC -DTECHNICAL_DATA -DTRY_EPG -DHAVE_DABSTICK -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/epg -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Isrc/input -Isrc/input/rawfiles -Isrc/input/wavfiles -isystem /usr/local/include -I/home/jan/rtl-sdr/include -Isrc/input/dabstick -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtXml -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o ofdm-processor.o src/ofdm/ofdm-processor.cpp
src/ofdm/ofdm-processor.cpp: In member function ‘virtual void ofdmProcessor::run()’:
src/ofdm/ofdm-processor.cpp:310:1: warning: label ‘Initing’ defined but not used [-Wunused-label]
 Initing:
 ^~~~~~~
src/ofdm/ofdm-processor.cpp:333:1: warning: label ‘SyncOnNull’ defined but not used [-Wunused-label]
 SyncOnNull:
 ^~~~~~~~~~
src/ofdm/ofdm-processor.cpp:362:1: warning: label ‘SyncOnEndNull’ defined but not used [-Wunused-label]
 SyncOnEndNull:
 ^~~~~~~~~~~~~
src/ofdm/ofdm-processor.cpp:408:1: warning: label ‘Block_0’ defined but not used [-Wunused-label]
 Block_0:
 ^~~~~~~
src/ofdm/ofdm-processor.cpp:436:1: warning: label ‘Data_blocks’ defined but not used [-Wunused-label]
 Data_blocks:
 ^~~~~~~~~~~
src/ofdm/ofdm-processor.cpp:464:1: warning: label ‘NewOffset’ defined but not used [-Wunused-label]
 NewOffset:
 ^~~~~~~~~
src/ofdm/ofdm-processor.cpp:494:1: warning: label ‘ReadyForNewFrame’ defined but not used [-Wunused-label]
 ReadyForNewFrame:
 ^~~~~~~~~~~~~~~~
g++ -c -m64 -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -fPIC -DTECHNICAL_DATA -DTRY_EPG -DHAVE_DABSTICK -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/epg -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Isrc/input -Isrc/input/rawfiles -Isrc/input/wavfiles -isystem /usr/local/include -I/home/jan/rtl-sdr/include -Isrc/input/dabstick -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtXml -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o ofdm-decoder.o src/ofdm/ofdm-decoder.cpp
src/ofdm/ofdm-decoder.cpp:282:7: warning: extra tokens at end of #endif directive [-Wendif-labels]
 #endif;
       ^
src/ofdm/ofdm-decoder.cpp:219:49: error: no ‘float ofdmDecoder::computeQuality(DSPCOMPLEX*)’ member function declared in class ‘ofdmDecoder’
 float ofdmDecoder::computeQuality (DSPCOMPLEX *v) {
                                                 ^
[... cut ...]
src/ofdm/ofdm-decoder.cpp:322:1: warning: label ‘toBitsLabel’ defined but not used [-Wunused-label]
 toBitsLabel:
 ^~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp:336:1: warning: label ‘handlerLabel’ defined but not used [-Wunused-label]
 handlerLabel:;
 ^~~~~~~~~~~~
src/ofdm/ofdm-decoder.cpp: At global scope:
src/ofdm/ofdm-decoder.cpp:246:5: warning: ‘cnt’ defined but not used [-Wunused-variable]
 int cnt = 0;
     ^~~
Makefile:797: recipe for target 'ofdm-decoder.o' failed
make: *** [ofdm-decoder.o] Error 1

JvanKatwijk, 2017-02-13:
You are absolutely right, my mistake

In the file "src/ofdm/ofdm-decoder.cpp" there is a function
"computeQuality" that has to be
enclosed by #ifdef HAVE_SPECTRUM ... #endif

I was playing around with some code to define some kind of quality
indicator and did not realize the function specification
(in the corresponding .h file was in a section enclosed by #ifdef
HAVE_SPECTRUM .. #endif

again, apologies

I'll upload a new version tonight, now I am in the middle of an editing
process
best
jan

closed 2018-02-12 by JvanKatwijk