JvanKatwijk/qt-dab #14: Build fail on Rawhide

russel, 2017-02-14:
I have Git master up-to-date as at 2017-02-14T07:53+00:00 and am using the CMake build. I am getting an error:
```
/home/russel/Repositories/Git/Forks/Qt_DAB/src/ofdm/ofdm-decoder.cpp:219:49: error: no ‘float ofdmDecoder::computeQuality(DSPCOMPLEX*)’ member function declared in class ‘ofdmDecoder’
 float ofdmDecoder::computeQuality (DSPCOMPLEX *v) {
                                                 ^
```

JvanKatwijk, 2017-02-14:
I know that one, my mistake, a function should have been enclosed in #ifdef
HAVE_SPECTRUM ...#endif.
Funtion is "computeQuality" in src/ofdm/ofdm-decoder.cpp"
Will be repaired later today

russel, 2017-02-14:
Splendid thanks.

closed 2018-02-12 by JvanKatwijk