JvanKatwijk/qt-dab #15: qt-dab does not install to expected place

russel, 2017-02-14:
Currently it seems, using the CMake build, that install happens to a place other than $prefix/bin.

JvanKatwijk, 2017-02-14:
I am not sure I understand, running CMake/make install delivers an
executable in the specified place ../linux-bin

j

russel, 2017-02-14:
It is that that place is non-standard and unexpected. When I run the build initialisation with:
```
cmake -DCMAKE_INSTALL_PREFIX=$HOME/Built <path-to-source>
```
then "make install" should install to $HOME/Built/bin. The default CMAKE_INSTALL_PREFIX is /usr/local I believe so the current hardwired location of ../linux_bin is unexpected to CMake users.

andimik, 2017-02-14:
Why do you expect it in built/bin?

I agree with Jan and cannot follow you as well. See readme file, where the path to the executable is mentioned.

russel, 2017-02-14:
This is about using CMake as the build tool. People using CMake expect the install location to be determined by the CMAKE_INSTALL_PREFIX variable.

JvanKatwijk, 2017-02-15:
I agree, it is the way CMake defines the world.
jan

closed 2018-02-12 by JvanKatwijk