JvanKatwijk/qt-dab #17: fft view

zygmund2000, 2017-02-17:
Hi

I have suggestion for fft view to change end lever form -40 to -30 when dabstic is used.

Regards

JvanKatwijk, 2017-02-17:
Can you be more specific?
j

zygmund2000, 2017-02-17:
Range of fft view, -40 is too large, there is only noise below -30,
    and I have question where in code can I change color of grid?

JvanKatwijk, 2017-02-17:
The Y scale is computed, based on the amount of bits your input device has.
With 8 bits you have - in principle - 48 db, that is what the scale tries
to express. You can locally change that by redefining the function
setBitDepth in the src/input/dabstick file.  If you let that function
return 6 than the bottom line of the scope will be app 30 db

Wrt color, that was a good one, stood for a long time on my list, but now
implemented:
band=BAND III
channel=11C
device=airspy
displaycolor=blue
latency=0
soundchannel=default

Just specify an internationally recognized color name in the :.ini" file

zygmund2000, 2017-02-17:
you can live it as is, but this color setting changes only background color not grid.

JvanKatwijk, 2017-02-17:
Now by setting displaycolor you set the background color, by setting
gridcolor you set the color of the grid.

I have to do some other things now, so the next question - if any - will
get a delayed answer
best
jan

closed 2018-02-12 by JvanKatwijk