JvanKatwijk/qt-dab #19: Make fails on qwt error

steeley, 2017-02-18:
Almost got all the QT/Qwt stuff compiled except ld fails looking for a file called -lqwt-qt5.
I can find no reference online to this file. What is it?

/usr/bin/ld: cannot find -lqwt-qt5
collect2: error: ld returned 1 exit status
Makefile:377: recipe for target 'linux-bin/qt-dab-0.998' failed
make: *** [linux-bin/qt-dab-0.998] Error 1

JvanKatwijk, 2017-02-18:
qwt is the qwt library. The naming is different on different installations.
Since you apparently do have qwt installed (at least the system does not
complain about missing declarations), it is somewhere on your system, maybe
just
named qwt
j

steeley, 2017-02-18:
agreed I was expecting qwt. I do have a proper qwt qt  install and the paths are all correct.
I've always found Qt to be a pain , so I guess  I'll   just use without the spectrum

andimik, 2017-02-21:
You could try to edit the qt-dab.pro file and others by adopting the path to qwt to your installation.

stefantalpalaru, 2017-03-04:
The library is sometimes installed as qwt6-qt5.so instead of qwt-qt5.so .

drusid, 2017-03-04:
Just edit the qt-dab.pro like this, I can only get it to compile using the sourceforge  qwt https://sourceforge.net/projects/qwt/files/qwt/6.1.3/

#correct for the correct path to the qwt6 library on your system
	LIBS		+= -lqwt
#	LIBS		+= -lqwt5

Wikinaut, 2017-03-20:
Re: https://github.com/JvanKatwijk/Qt-DAB#ubuntu-linux

On Linux Mint 18.1 (Ubuntu 16.04), I had to add

`sudo apt install libqwt-qt5-dev`

to fix this issue.

Wikinaut, 2017-03-20:
see #26 (doublure)

andimik, 2017-05-05:
Is there any problem if more than one version is installed?  On my rpi3 under Jessie I try to get the spectrum run.

But I have tried qwt5, qwt 6.3.1 without success. 

Should I uninstall 6.3.1?

JvanKatwijk, 2017-05-05:
Hi Andreas

Just back, so a serious backlog in the mail. If I understand you correctly,
you have more than one Qt /qwt version installed?
I am not sure what the effect is when you compile against version qwt6.2
and run with 6.3. I know that in case of
compiling against 6 and running with 5 makes your program crash.

best
jan

closed 2018-02-12 by JvanKatwijk