JvanKatwijk/qt-dab #20: cmake Error

Roughtrade, 2017-03-01:
I have the following error using cmake:
cmake -DDABSTICK=ON -DRTLTCP=ON 
`[ 76%] Building CXX object CMakeFiles/qt-dab-0.998.dir/optional-scope/spectrum-handler.cpp.o
/home/snoopy/qt-dab/optional-scope/spectrum-handler.cpp: In constructor ‘spectrumhandler::spectrumhandler(RadioInterface*, QSettings*, RingBuffer<std::complex<float> >*, RingBuffer<std::complex<float> >*)’:
/home/snoopy/qt-dab/optional-scope/spectrum-handler.cpp:88:19: error: ‘class QwtPlotCurve’ has no member named ‘setOrientation’
  spectrumCurve -> setOrientation (Qt::Horizontal);
                   ^
/home/snoopy/qt-dab/optional-scope/spectrum-handler.cpp: In member function ‘void spectrumhandler::ViewSpectrum(double*, double*, double, int32_t)’:
/home/snoopy/qt-dab/optional-scope/spectrum-handler.cpp:202:19: error: ‘class QwtPlotCurve’ has no member named ‘setSamples’
  spectrumCurve -> setSamples (X_axis, Y1_value, displaySize);
                   ^
make[2]: *** [CMakeFiles/qt-dab-0.998.dir/optional-scope/spectrum-handler.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/qt-dab-0.998.dir/all] Fehler 2
make: *** [all] Fehler 2
`

JvanKatwijk, 2017-03-02:
Hi

Your error message contains references to
optional-scope/spectrum-handler.cpp. which is a little strange since you
did not specify -DSPECTRUM=ON.
It seems to indicate that somehow SPECTRUM is on.

If SPECTRUM is defined, then qwt will be referred to. The method
setOrientation is available in both qwt5 and qwt6/

Pls send me the CMakeLists.txt file that you were using and give me some
specs of the sw installation (i.e. qtXX, qwtXX)

best
jan

I

Roughtrade, 2017-03-02:
Hi, jan i did it again. But i have changed libqwt5-qt4-dev to libqtw-dev 6.0.0-1.2

`cmake .. -DDABSTICK=ON -DRTLTCP=ON -DSPECTRUM=ON
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'fftw3f >= 3.0'
--   found fftw3f , version 3.3.3
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- checking for module 'portaudio-2.0'
--   found portaudio-2.0, version 19
-- Performing Test PORTAUDIO2_FOUND
-- Performing Test PORTAUDIO2_FOUND - Success
-- Found PORTAUDIO: /usr/include  
-- Found zlib: 
-- Found LibSndFile: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Found LibSampleRate: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- checking for module 'librtlsdr'
--   found librtlsdr, version UNKNOWN
-- Found librtlsdr: /usr/local/include, /usr/local/lib/librtlsdr.so
-- QWT Version: 6.0.0
-- Found Qwt: /usr/lib/libqwt.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/snoopy/qt-dab/build
`
make 

`[ 92%] Building CXX object CMakeFiles/qt-dab-0.998.dir/includes/backend/data/moc_data-processor.cpp.o
[ 93%] Building CXX object CMakeFiles/qt-dab-0.998.dir/src/input/rawfiles/moc_rawfiles.cpp.o
[ 94%] Building CXX object CMakeFiles/qt-dab-0.998.dir/src/input/wavfiles/moc_wavfiles.cpp.o
[ 95%] Building CXX object CMakeFiles/qt-dab-0.998.dir/moc_radio.cpp.o
[ 96%] Building CXX object CMakeFiles/qt-dab-0.998.dir/src/input/dabstick/moc_dongleselect.cpp.o
[ 97%] Building CXX object CMakeFiles/qt-dab-0.998.dir/src/input/dabstick/moc_dabstick.cpp.o
[ 98%] Building CXX object CMakeFiles/qt-dab-0.998.dir/src/input/rtl_tcp/moc_rtl_tcp_client.cpp.o
[ 99%] Building CXX object CMakeFiles/qt-dab-0.998.dir/optional-scope/moc_spectrum-handler.cpp.o
[100%] Building CXX object CMakeFiles/qt-dab-0.998.dir/includes/scopes-qwt6/moc_iqdisplay.cpp.o
Linking CXX executable qt-dab-0.998
[100%] Built target qt-dab-0.998`

;./qt-dab-0.998 
Speicherzugriffsfehler (Speicherabzug geschrieben)

dmseg

[593543.910587] qt-dab-0.998[8451]: segfault at 7fcbab0fdcd0 ip 00007fcba77cbdfc sp 00007fff86a303e0 error 7 in libQtGui.so.4.8.6[7fcba7614000+a66000]

Roughtrade, 2017-03-03:
Thank you !!!!!! 
every thing works fine now.

closed 2017-03-03 by Roughtrade

JvanKatwijk, 2017-03-03:
OK, good to hear. I would say: have fun

best
jan