JvanKatwijk/qt-dab #21: Slideshows don't work in Slovenia and Slovakia

andimik, 2017-03-06:
I found out that the slideshows in qt-dab and eti-stuff don't work at all in Slovenija and in Slovakia, whereas dablin_gtk or XPADxpert (and the DABPlayer from Andreas Gsinn under Windows) show them. 
[image]

BTW: In Vienna it worked under qt-dab well on the same Laptop, see screenshot: .
[image]

According to XPADxpert it's in the same subchannel.

See the (zipped) eti-file http://andimik.bplaced.net/temp/10d_lang2.eti.zip , please load in eti-backend, tune to the English speaking "Slo Int Radio SI" (other programs didn't have slideshows at that time and the sample is too short).

andimik, 2017-03-21:
BTW in welle.io the (Slovene) slides work as well.

JvanKatwijk, 2017-03-21:
Hi Andreas

I'll look into it, You know of course that welle.io uses the full qt-dab
dab engine (It basically provides a different interface and I noticed that
the sound out is different. The dab decoding uses the qt-dab files) I had a
quick look and he uses the pad and mot handling from dablin. (welle.io does
not handle dab data, only program associated date).
I ran a quick test on the file you sent me a couple of weeks ago, and I
found this. I know there are problems under Windows, that
has to do with the Qt engines.

Anyway, if you can provide me with some large files with raw data where
qt-dab cannot extract slides while the dablin software can, it will be
really helpful to address this issue

best
jan

andimik, 2017-03-24:
Please see https://github.com/AlbrechtL/welle.io/issues/30#issuecomment-288872557

In this raw file you should see slideshow on 2 services:

on SId 0x2248, 'RTE GOLD' (subchannel 9)
[image]

and on SId 0x2202, 'RTE 2FM' (subchannel 2)
[image]

They only appear in dablin_gtk (after I have exported the ETI frames with eti-frontend) and in XpadXpert.

Welle.io, qt-dab and eti-backend do not show them ...

andimik, 2017-03-29:
Now it's a bit better. Some stations now show slideshows. Playing the Irish raw file I got it for 2FM:

> we found for RTE 2FM          serviceId 2202
we have now DAB
going to show picture slide2509.jpg

But still no slideshow on RTE GOLD:

> we found for RTE GOLD         serviceId 2248
we have now DAB
End of file, restarting
End of file, restarting
End of file, restarting

Same in Slovenia, Slideshow not shown (see screenshot above, this is the same file I investigated) here:

> we found for Slo Int Radio SI serviceId 933A
we have now DAB+
End of file, restarting

but success here:

> we found for Radio Ognjisce   serviceId 942F
we have now DAB+
going to show picture

JvanKatwijk, 2017-03-30:
I know it is not perfect yet. There is one case where the incomping
packages fail on the crc test. If I skip the crc test you see most of the
picture, it is the very last segment that cripples the data.
To be continued this weekend
best
jan

basicmaster, 2017-03-31:
Regarding RTE GOLD it looks like that between the MOT Header (with correct CRC) and a MOT Body repetition (with correct CRC) there are further transmissions of the same MOT Headers/Bodys, but all with wrong CRC due to e.g. missing AUs. Maybe this is the reason here.

JvanKatwijk, 2017-04-01:
Wrt RTE GOLD, reading the file (and continuing after a failing crc), this
happens

we have now DAB
crc failed grouptype 4, segmentNumber 0, lastFlag 1
crc failed grouptype 3, segmentNumber 0, lastFlag 1
crc failed grouptype 4, segmentNumber 0, lastFlag 1
going to show picture slide11.jpg�
Corrupt JPEG data: premature end of data segment
cannot write file slide11.jpg�
crc failed grouptype 3, segmentNumber 0, lastFlag 1
crc failed grouptype 4, segmentNumber 0, lastFlag 1
crc success grouptype 3, segmentNumber 0, lastFlag 1
ALSA lib pcm.c:7963:(snd_pcm_recover) underrun occurred
End of file, restarting

As can be seen from the partial picture, the last part of the picture is
corrupted. Note that the picture is encoded in a
single msc group (so the complete picture and description is in a single
group3 group4 combination), It is not shown here,
but the size of the container for the group 4 block is OK, slightly larger
than the size of the block.

basicmaster, 2017-04-01:
Also when the CRC is checked and failing DGs are discarded, (exactly) one picture should be displayed. I just took a look into the file and it happens that here the MOT body is transmitted (uncorrupted) before the MOT header:

```
MOTDecoder: mot_len:    39, CRC: OK		transport_id: 13829	file: slide2505.jpg
MOTDecoder: mot_len:    39, CRC: wrong
MOTDecoder: mot_len:  7926, CRC: OK		transport_id: 13835
MOTDecoder: mot_len:    39, CRC: wrong
MOTDecoder: mot_len:  7926, CRC: OK		transport_id: 13835
MOTDecoder: mot_len:    39, CRC: OK		transport_id: 13835	file: slide2511.jpg
MOTDecoder: mot_len:    39, CRC: OK		transport_id: 13829	file: slide2505.jpg
```

andimik, 2017-06-07:
Newer versions show them, but it's still not running smoothly. 

Anyway I'll close it but maybe basicmaster could help you in this matter.

andimik, 2018-05-15:
Jan, could you try to improve the code for MOT handling according to @basicmaster 's proposals.

https://github.com/JvanKatwijk/qt-dab/issues/21#issuecomment-290838912
https://github.com/JvanKatwijk/qt-dab/issues/21#issuecomment-290945383

The MOT icon is very often "blinking" and slideshows are not shown or it takes very, very long. All my other tools show slideshows when the file has been received.

Maybe you could use code from welle.io?

JvanKatwijk, 2018-05-15:
I'll look into it, right now I am trying to figure out how a hackrf is to
be interfaced
So, it might be next week

andimik, 2018-05-26:
With https://github.com/JvanKatwijk/qt-dab/commit/1874dd9bebffd5c42ff13ca655e30e9f024a9c9d it's the same, still not all slides are shown.

I've tried it several times with the same rawfile (in my case one of the radio programs has 2 alternating slides), sometimes the slides appear, but in lots of the cases they still don't and MOT is blinking like before.

JvanKatwijk, 2018-05-26:
> With 1874dd9

can you send me a link to that file then I can have a look.
The blinking is because each time a new message starts I switch it off, and
only switch it
on when everything is/seems OK

wrt the naming: I think it is OK now

andimik, 2018-05-26:
If you still have mine from Slovenia, then use `Slo Int Radio SI`, they have never changed their stream since their start in 2016.

zygmund2000, 2018-08-17:
Hi,

 Lastest version of qt-dab don't display slides from ODR tools, other receivers working fine.

Regards

JvanKatwijk, 2018-08-17:
Is the picture written onto a file?

andimik, 2018-08-17:
Which OS?

JvanKatwijk, 2018-08-17:
It is Linux

I just ran a quick test and notices that a version from may showed a
picture and the current one does not

The interesting this is that the saved versions of the pictures, generated
with the two instances, are exactly the same (so differences when comparing)
The code that should display the picture is also the same in the may and
current version, so I have to puzzle a little

JvanKatwijk, 2018-08-17:
I did not look at windows yet, my windows system is a mess

For Linux:

if/when you use the Qt-DAB appImage then the pictures do not show, if you
compile your own version they do
The AppImage is made in an old Ubuntu environment. Why? it does contain
most of the libraries needed, however
it depends on some generally available libraries and to prevent that the
AppImage has to rely on older "general"
libraries, it is built on a pretty old system.

2018-08-17 11:42 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> It is Linux
>
> I just ran a quick test and notices that a version from may showed a
> picture and the current one does not
>
> The interesting this is that the saved versions of the pictures, generated
> with the two instances, are exactly the same (so differences when comparing)
> The code that should display the picture is also the same in the may and
> current version, so I have to puzzle a little
>
>
>
>> Which OS?
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980
> +31 (0) 628260355

andimik, 2018-08-17:
Could you check if jpg and png behavior the same?

zygmund2000, 2018-08-17:
I don's see pictures in specified directory picturesPath=/home/user

JvanKatwijk, 2018-08-17:
The default is /tmp/qt-pictures

Btw, I assume you use a Linux version

JvanKatwijk, 2018-08-17:
> Yes Linux version (Manjaro), I changed default to /home/user but
> /tmp/qt-pictures also no images.
> [image: mot]
> <https://user-images.githubusercontent.com/14185298/44287213-499d1880-a26c-11e8-8e5e-5bc481dbce90.png>

did you compile yourself or did you use the appimage
is manjaro ubuntu based?

zygmund2000, 2018-08-17:
I compiled from git, manjaro is arch based.

andimik, 2018-08-17:
Under kubuntu (compiled with qmake and make) they both work: png and jpg

JvanKatwijk, 2018-08-18:
What is on your system the behavious of you use an AppImage (see the
releases section of the github rep)

zygmund2000, 2018-08-18:
Also no slides, please check it with DAB generated by ODR tools, maybe bug is not in qt-dab.

andimik, 2018-08-18:
Could you share an eti file?

andimik, 2018-08-18:
I can't confirm that it should be an issue with ODR. I tried various eti streams from satellite and made a raw and opened it in qt-dab. 

They all work. 

But, you are right, the initial bug is still unsolved. The MOT is blinking and slides are shown rarely, sometimes after several replays. Basicmaster gave a hint, so Jan could check the code again.

zygmund2000, 2018-08-18:
This issue is difficult to track because slides work fine on eg. LG Stylus and welle and dablin but not with qt-dab today, some older versions of qt-dab sometimes show images after long time.
IMHO ODR-padenc has something different than commercial encoders because this muxes which use it have always images on every my receiver including qt-dab also.

JvanKatwijk, 2018-08-19:
With the recording I have (10D_Slovenia from 2017-12-26) there are no
differences in output
with different qt-dab versions (from april until right now)

It would really be helpful to have some recording that in your context
shows the
difference in behaviour.

zygmund2000, 2018-08-19:
I'll try to save eti file or I can share as live stream if you wish

JvanKatwijk, 2018-08-20:
For me easiest would be to get a pointer to a raw input file, then I can
trace all steps and see where things go wrong

JvanKatwijk, 2018-08-20:
> raw IQ from air?

Yes, just a raw input file (either 8 bits raw or made by qt-dab, a wav file)
In the mean time I found one thing that might have caused a problem, it is
corrected and put in the repository,

zygmund2000, 2018-08-20:
Still nothing, I sent IQ to email.

zygmund2000, 2018-08-20:
Now I see slides, can you explain why others muxes work before this change?

JvanKatwijk, 2018-08-20:
That is pretty complex. It has to do with detection of erroneous frames. It
turned out that in some cases an msc datagroup was erroneously
rejected. That followed from the handling of an error that occurred with an
English DAB service, I then strengthened the test to
check whether or not a datagroup was (supposed to be) legal

closed 2020-09-27 by JvanKatwijk