JvanKatwijk/qt-dab #22: dab-cmdline + dablin

zygmund2000, 2017-03-07:
Hi,

Is possible to use dab-cmdline with this https://github.com/Opendigitalradio/dablin together? Or in future?

Regards

JvanKatwijk, 2017-03-07:
> Hi,
>
> Is possible to use dab-cmdline with this https://github.com/
> Opendigitalradio/dablin together? Or in future?
>
> Regards

Suggestions to extend dab-cmdline are more than welcome. As you know, the
dab handling is now completely done in a separate library, just to
investigate the possibility of combining it with other items

to be continued
best
jan

andimik, 2017-03-08:
Well, eti-stuff is working with dablin_gtk (basically), but the sound is not good, and I have dropouts ...

`./eti-frontend -O - | dablin_gtk`

> SDLOutput: silence started...
> SDLOutput: silence ended (115200 bytes)
> SDLOutput: silence started...
> SDLOutput: silence ended (117760 bytes)
> SDLOutput: silence started...
> SDLOutput: silence ended (199680 bytes)
> SDLOutput: silence started...
> SDLOutput: silence ended (101376 bytes)
> SDLOutput: silence started...

andimik, 2017-04-18:
Jan, I found an older version from 14th Jan 2017, where these dropouts do not happen when streaming to stdout and to dablin_gtk

JvanKatwijk, 2017-04-19:
The eti frontend is a difficult program, since it has too many options.
The "startindex = -3" you referred to indicates that the synchronization
was lost. eti generation assumes a correctly build
(and maintained) internal structure.

For a number of reasons I prefer the command line version, much, much
simpler to maintain. My suggestion would be to
delete the eti-frontend completely and focus on the library version.

Can you send me a link to an inputfile where you observe the difference
between the 14 january version and the current version,
that could be very helpful. In a few tests I ran this morning the versions
did not show differences (but then, with lousy input, so
there were provable dropouts due to loss of synchronization with both)
best
jan

closed 2018-02-12 by JvanKatwijk

basicmaster, 2018-02-12:
@zygmund2000:
Just for the sake of completeness: Since a while it is possible to use `eti-cmdline` together with DABlin (like it was already possible with `dab2eti`).