JvanKatwijk/qt-dab #23: dabstick autogain value not restored

nottledim, 2017-03-10:
The "Autogain" setting for dabstick is saved in ~/.qt-dab.ini but is not restored on next invocation of qt-dab.

The value saved looks like: "autogain=autogain on"

The software is running on Debian GNU/linux (testing) amd64 with Nooelec nesdr.  qt-dab was downloaded 9th Mar 17.

N.B. It might be helpful to default this setting to "on" as the receiver does very little with autogain off.

JvanKatwijk, 2017-03-12:
The "autogain" is now maintained between invocations.
Wrt the default setting: also done
(I must admit, my dabstick is mechanically pretty weak, it is not my
favorite device, I seldom use it)

best
jan

nottledim, 2017-03-12:
On 03/12/17 13:16, JvanKatwijk wrote:
> The "autogain" is now maintained between invocations.
> Wrt the default setting: also done

Something's gone wrong, it doesn't work, neither the restore previous setting
or default.  Also it now restores value of gain incorrectly.

> (I must admit, my dabstick is mechanically pretty weak, it is not my
> favorite device, I seldom use it)

The Nooelec is well made but my cheap ebay thing is, er, a bit cheap.  It does
work though.

Dick

nottledim, 2017-03-12:
I think it's that you've written combo_gain instead of combo_autogain on line 185 of dabstick.cpp

JvanKatwijk, 2017-03-12:
Oops, you are absolutely right. The error was that I figured that "autogain
on" and "autogain off" was recognized when being read. However, the string
read was "autogain", the change is that an underscore is added.
The fact that the gain was changed is related, but difficult to explain in
a few words

With my dabstick it works
best
jan

closed 2017-03-14 by JvanKatwijk