JvanKatwijk/qt-dab #25: Prerequisites to get sound

andreygursky, 2017-03-14:
Jvan, thanks for qt-dab!

I've managed to compile it on Debian Stretch (testing):
* Qt4:
  * install ```libqwt-dev```
  * qt-dab.pro: ```LIBS += -lqwt```
  * ```QT_SELECT=4 qmake qt-dab.pro```
  * ```make```
* Qt5:
  * install ```libqwt-qt5-dev```
  * qt-dab.pro: ```LIBS += -lqwt-qt5```
  * ```QT_SELECT=4 qmake qt-dab.pro```
  * ```make```

(just in case someone would need for reference)

Using dabstick, channels are found but I get no sound. RAW data can be saved, but not audio file. Technical info looks like:
[image]
What are minimal prerequisites to decode/play audio? (And what do these 4 horizontal bars mean?)

Thanks,
Andrey

andimik, 2017-03-14:
Your reception or your settings are too poor for sound. All 4 bars have to be 100%. What about other channels (5C, Bundesmux)?

Try to play with the gain settings and check the spectrum. It shall look like the screenshot in https://github.com/JvanKatwijk/qt-dab#introduction

andreygursky, 2017-03-14:
Oh, that's bad. Other channels look similarly. They are all on 11D. Perhaps to week antenna for DAB.

andreygursky, 2017-03-15:
I was pointed to a short description of the bars via tool-tip.

zolant, 2017-05-19:
hello,
can't compile after yesterday update. Previously (5 days ago) used to work.
/tmp/ccoKIXXi.ltrans2.ltrans.o: In function `RadioInterface::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)':
<artificial>:(.text+0x686): undefined reference to `RadioInterface::showCoordinates(float, float)'
collect2: error: ld returned 1 exit status
Makefile:395: recipe for target 'linux-bin/qt-dab-0.999' failed

Just curious what it could be, where I'm doing wrong? Doing this on odroid (clone of Raspberry)

JvanKatwijk, 2017-05-19:
You are absolutely right. In the file includes/ofdm/ofdm-processor.h, near
last line it should read

#ifdef  TII_COORDINATES
        void            showCoordinates         (float, float);
#endif

So, I forgot to embrace the specfication in "ifdef" ... endif pair

is repaired in the github

FYI, the TII_COORDINARES is an attempt to extract location information from
the DAB signal. It takes a few FFT's extra once a minute

closed 2017-07-10 by JvanKatwijk