JvanKatwijk/qt-dab #27: Channel is not updated in ~/.qt-dab.ini

HawkingJan, 2017-03-20:
I'm using qt-dab with an dabstick on my RaspberryPi.
If I start the gui, change the channel, listen to some music and then quit the program by clicking the "quit" button I get

```
ofdm decoder is closing down now
Termination started
closed
Segmentation fault

```

I don't know if the segmentation fault is the reason why the channel in the ~/.qt-dab.ini is not being updated. Of course I can update the chanel in the file manually, however I would prefer if the last used channel is automatically saved. The other values (device and soundchannel) were properly updated.

andimik, 2017-03-21:
You are not the only one getting segment fault when closing the program. But I have not tested if the ini file is written or not.

JvanKatwijk, 2017-03-21:
I assume you are using Qt5. On my RPI 2 I am using Qt4 and I do not have
the problem with the crash on exit.
With Qt5 there is a problem, I do have that on my laptop. According to
sources on the internet it is a known problem, and in my case it appears
after having left my code.
However, the channel is saved properly both on the Qt-5 version on the
laptop and the Qt4 version on the RPI 2.

Pls try the latest version of the github, I added a "sync" for the ini
file, right after saving some values.
Jan

HawkingJan, 2017-03-21:
Thanks for the quick reply. 

 > I assume you are using Qt5. 

I read about the Qt4 vs. Qt5 in the description of the project and wanted to use Qt4.
Therefore I replaced
`LIBS            += -lqwt-qt5`
by 
`LIBS            += -lqwt`
before compilation. I'm not sure if this is the correct way or if other changes (besides the removal of the
```
CONFIG         += sdrplay
CONFIG         += rtl_tcp
CONFIG         += airspy
```
block) should have been done?

>  Pls try the latest version of the github

Now the channel is being updated but the segmenation fault remains (but does not "hurt" much as far as I can see).  Maybe you can comment on how to change to QT4?

JvanKatwijk, 2017-03-21:
You can safely remove the sdrplay, airspy and rtl_tcp. No problem
Qwt is used for generating the spectrum display. You do not need it is you
comment out
#CONFIG += spectrum

If you do not have qwt (and therefore do not have problems, commonly
associated with qwt), you just have to install Qt4.
with apt-cache search Qt4 you can identify the packages required

best
jan

HawkingJan, 2017-03-21:
I think there is a misunderstanding. 
I know that qwt is used for the spectrum display (which I want to see and is displayed).
I'm just not sure if I'm using QT4 or QT5 (I installed qt4-qmake, qt4-default libqt4-opengl-dev and replaced LIBS += -lqwt-qt5 by LIBS += -lqwt before compilation but I'm not sure if this is correct/sufficient because you assumed that my segmentation fault is related to using QT5)

Best
Jan

JvanKatwijk, 2017-03-21:
Well, if you run qmake-qt4 with a qwt lib for qt5, or the other way around,
qmake-qt5 with a qwt lib for qt4 your program will crash before showing
anything. The difficulty - with all debian based systems - is to identify
the correct qwt for the qt you use.
I am using the settings - and naming- as in the ".pro" file, and my
installation is qmake-qt4/qmake
Jan

HawkingJan, 2017-03-21:
If I use the settings as in the default qt-dab.pro file (only removed sdrplay, rtl_tcp and airspy) and use
qmake qt-dab.pro && make
I get during the make step
```
/usr/bin/ld: cannot find -lqwt-qt5
collect2: error: ld returned 1 exit status
Makefile:242: recipe for target 'linux-bin/qt-dab-0.998' failed
make: *** [linux-bin/qt-dab-0.998] Error 1
```
Therefore I replaced LIBS += -lqwt-qt5 by LIBS += -lqwt (the later is already in the qt-dab.pro file but commented out)

I now tested qmake-qt4 instead of qmake and still got the same error.
Anyway, the problem with not saving the channels is solved and as far as I can see the segmentation fault is only a minor problem and I think that the ticket can therefore be marked as solved.

JvanKatwijk, 2017-03-21:
Crashing of a program is never OK. In my RPI the software was built with
qmake-qt4 abd does not crash on exit.
Any clue of what is happening on program termination is welcome to have
this further investigated

best
jan

HawkingJan, 2017-05-21:
I just wanted to let you know that the segmentation fault still exists in the latest (0.999) version:

```
ofdm decoder is closing down now
.. end the radio silences
Segmentation fault
```
I'm not sure if this is the reason why the [dabstickSettings] in the ~/.qt-dab.ini file is not being updated to the latest settings (offsets, ...).

JvanKatwijk, 2017-05-21:
Pity that there is a segmentation error in your environment, on my RPI (2),
my fedora and my Ubuntu boxes that does not happen.

The settings are saved well before the segmentation error happens, and -
according to the output you give - nothing further happens to the
QSettings. Again I cannot replay that behaviour on my development box. If I
let the program crash just after the point where the message

. end the radio silences

then the settings are saved.

Where I am now I do not have a dabstick available, I'll test again
with dabstick later this week.

One thing you could try is to add the line given below to
        rtlsdrSettings  -> beginGroup ("rtlsdrSettings");
        rtlsdrSettings  -> setValue ("externalGain", theGain);
        rtlsdrSettings  -> setValue ("autogain",
                                              combo_autogain -> currentText ());
        rtlsdrSettings  -> setValue ("f_correction",
                                              f_correction -> value ());
        rtlsdrSettings  -> setValue ("KhzOffset",
                                              KhzOffset -> value ());
        rtlsdrSettings  -> endGroup ();

this one

      rtlsdrSettings -> sync ();

in the destructor of the the rtlsdrHandler, in the file rtlsdr-handler.cpp

HawkingJan, 2017-05-21:
Thanks for your quick reply. 
I removed the part about the settings not beeing saved from my previous post. 
Somehow I assumed that these settings would represent the  "small" and ”large” freqency offset shown in the main window and that change while running. I thought that these settings would be saved so that for the next run the ideal start settings would exist. I simply missed that there is an extra window where I can explicitly set these parameters. If I do so they are however still not saved.

I now changed the rtlsdr-handler.cpp as you said and ended up having a still unchanged [dabstickSettings] part and a new [rtlsdrSettings] part (with actual values different from 0) in the ini file. Is this how it should be?

JvanKatwijk, 2017-05-21:
I am not certain what you mean. The "coarse" and "fine" frequency offset
are not stored. Device dependent settings, i.e. the ppm and
user defined offset should be saved
[rtlsdrSettings]
KhzOffset=0
autogain=autogain_on
externalGain=364
f_correction=0

This is saved. The f_correction is meaningless (used to be meaningfull some
time ago), and the ppm-setting is not saved yet.
So, the useful things that are saved are the autogain setting and the gain
setting.

dabstickSettings is old, some time ago I started unifying names and so
dabstick was renamed in rtlsdrHandler. Yo can remove the section
"dabstickSettings" from the ini file without any problem

HawkingJan, 2017-05-21:
Just out of interest, wouldn't it be beneficial if the "coarse" and "fine" frequency offsets would be saved (as starting values for the next run)?

 > and the ppm-setting is not saved yet.

I used the ppm setting to test if the settings are begin saved. 
I now changed the "gain" value, however also this value is not beeing saved (only if I change the code as you said to).
Maybe you can add the line " rtlsdrSettings -> sync ();" in the next version (and include the saving of ppm values)?

JvanKatwijk, 2017-05-22:
I'll look into it later this week. I think I need to create a situation as
with your setup, such that it crashes

JvanKatwijk, 2017-05-22:
Two comments
a. the ppm offset is saved, it just is/was not named ppm offset
b. thinking on saving/restoring the coarse (and fine) frequency offset:
both are computed and handled in software. The frequency corrections are
done in software, they do no affect the frequency setting of the attached
device.
Fine frequency correction - i.e. in Mode 1 between -500 .. 500 Hz is pretty
dynamic, if you wave your hand around the antenna
you will see that changed. Coarse correction, i.e. the steps of 1 kHz do
not change much during operation , they are computed
when not in sync. On program start up the coarse offset is computed which
may take some time. I'll have to do some tests to see what the effect from
saving the coarse offset (again, per type of device) will be.

What I wold like to get is the configuration setting that you are using,
that allows me to create a version as much as possible similar to yours, in
order to see whether of where the segmentation violation on termination
occurs.

best
jan

2017-05-22 7:41 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> I'll look into it later this week. I think I need to create a situation as
> with your setup, such that it crashes
>
>
>
>> Just out of interest, wouldn't it be beneficial if the "coarse" and
>> "fine" frequency offsets would be saved (as starting values for the next
>> run)?
>>
>> and the ppm-setting is not saved yet.
>>
>> I used the ppm setting to test if the settings are begin saved.
>> I now changed the "gain" value, however also this value is not beeing
>> saved (only if I change the code as you said to).
>> Maybe you can add the line " rtlsdrSettings -> sync ();" in the next
>> version (and include the saving of ppm values)?
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

HawkingJan, 2017-05-22:
What information do you need to have?
I'm using Raspbian Jessie as a distribution. Starting from the default qt-dab.pro I comment out the lines 

```
CONFIG         += sdrplay
CONFIG         += rtl_tcp
CONFIG         += airspy
CONFIG         += elad_s1
```
so that only dabstick is left as a device. 

 And switch from `LIBS            += -lqwt-qt5` to `LILBS            += -lqwt.
Then I run 
```
qmake qt-dab.pro
make
```
I've gcc version 6.2 installed.
`

JvanKatwijk, 2017-05-23:
I had the RPI 2 run for a couple of hours and did not suffer from dropouts,
other than when there were frame errors. The average load I get is between
60 and 70 % (but I am running headless using an ssh connection). I do not
have Spectrum activated, since the connection is not fast enough. If/when
you have dropouts can that be related to CPU load? do you have Spectrum on?

Furthermore, I had some times crashes when terminating, I think I managed
to address that, tried a couple of hours

best

HawkingJan, 2017-05-23:
I can confirm that the segfault is gone

```
.. end the radio silences
back in main program
It is done

radioInterface is deleted
```
 and in my test the .qt-dab.ini is updated successfully.
So I think this ticket can be closed. I'm still having lots of dropouts. I will post an addition on the ticket that is related to the dropouts so that we can discuss this issue there.

However, I've one more question/suggestion regarding a faster start. Maybe you can give a quick reply here.
In addition to think about saving the coarse offset at exit, would it be possible to automatically start scanning for stations after start up of the program (at the moment one has to click on "start" and I wonder why this is necessary).

JvanKatwijk, 2017-05-24:
One of the undocumented features is that when you set "autoStart=1" in the
ini file, you do not have to push the start button

The need for the startbutton follows from the  many settings (in earlier
version) you might want to do before starting: selecting a mode, a band, a
channel and an input device

HawkingJan, 2017-05-30:
Thanks. "autoStart=1" works exactly as I wanted. Maybe you could think about adding the default (autoStart=0) to the initial ini file (together with a comment that explains the usage). This should be a simple change (compared to adding a checkbox to the gui) and would be a good indicator for the more advanced user that such a feature exists. 
Is there also an undocumented feature to save the last used radio station (to avoid problems with spaces in the names maybe the station id instead the station name) to the ini such that the station is automatically selected when it is found?

JvanKatwijk, 2017-05-31:
autoStart=0 will be added to the ini file. wrt saving the program name, not
built in.
I just ran a quick test and as I feared, it would involve setting some form
of timer, since recognizing that an ensemble is available
does not imply that the descriptive data for all programs is already in.
having a timer complicates things since the timer mist be switched off
whenever the user either selects a program him/herself
or selects another channel or selects another band or selects another mode.

In the command line version that was petty simple, since the state of the
started program is not changed from outside.

I really have to think about it

HawkingJan, 2017-05-31:
Thanks for your quick reply.

I understand that using a timer would be complicated but I don't understand why a timer is needed at all. 

Here my naive concept (not even pseudo-code):

At startup:
`read lastusedstation from ini`

In the function that refreshes the station list:
`if newfoundstation = lastusedstation then startplaying(newfoundstation)`

In the function that gets called when a user actively clicks on a station on the station list:
`lastusedstation = clickedstation`

At exit:
`update lastusedstation in ini`

But I don't know the code so most likely I missed some details.

JvanKatwijk, 2017-06-01:
A timeout is needed to let the system collect the descriptive data for the
programs in the ensemble. Having the name does not mean
having all the data for it, that may take handling of a few DAB frames
more  (the normal order is that the name is first, then additional data)

So, on entering the name in the namelist there is no garantee that the
program can be decoded.
I tried this setup yesterday as well.

The solution is to add a timer - in case there is a "lastusedstation" -
wait and then get the data for the program.
The programmatical complication is that in a dozen places that are
activated by the GUI you have to switch of the
timer and do other things.

Further unsolved issues are: what if someone writes in the ini file the
name of a non existent program? what if the device attached
does not support channels (e.g. files)? each issue by itself is fairly easy
but all those small things complicate the handler software.

The "specified station" approach is implemented in the cmdline program.
That is fairly easy since there is no further interaction
and if it does not work you can just stop the program.

Btw as a quick hack I added that if you specify in the ini file the
pictures="" , i.e. an empty path for the pictures, then the
pictures will not be shown. Writing this, it is a cleaner solution to just
add a field, something like "showPictures=0" or 1

Today I do not have time anymore, but that is a simple approach that I'll
add.

HawkingJan, 2017-06-02:
Thanks for doing all your tests.

Regarding the ppm value. If understand you correct it somehow affects the frequency.  I didn't find much information about it (in the pdf manual it is mentioned but without explanation). Can you provide some background? Especially what it does, how to find a reasonable setting (and save it)...

JvanKatwijk, 2017-06-03:
PPM offset (Parts per Million) is a value with which the oscillator in the
input device will be changed. This is merely a firmware/hardware
device issue for which the library of that device provides an entry.
Although the "error" in the frequency generated by the oscillator is
influenced by a lot of things (e.g. temperature), when using in a "home"
environment, the "error" is relatively constant. So, once you
know the error the ppm offset can be considered a "constant" to be used in
the setup of the device.

Changing the ppm when running the software is possible, the effect must
show in the computation of the coarse frequency offset (which is computed
over the signal when not in sync)

Hope this answers your question

The manual is completely outdated and partly replaced by tooltips. The
question with a manual is always the audience you target (i.e. people who
want to understand how to build the system, how it really works, and the
ones who only want to know which button to press)

HawkingJan, 2017-06-05:
Sorry for asking stupid follow up questions: Besides a faster start/switch of channels, is there a difference weather the frequency is corrected by the ppm setting at the start or at run time by the calculated coarse frequency? To get a good ppm value one should vary the value such that the coarse frequency offset is (close to) 0?
In an other post you mentioned ppm offset = 84 as an example. Do you have some experience what range of values on should try or does this strongly depend on the dabstick model?

HawkingJan, 2017-06-05:
And I just got an other segmentation fault error when closing the program (compiled after your fix 13 days ago)

```
ofdm decoder is closing down now
.. end the radio silences
Segmentation fault

```

I don't know hat I did different to the case were it worked (see some posts above). Are there any more details I should provide?

JvanKatwijk, 2017-06-06:
OK, when I am back from traveling (in july) I'll try to figure out which
thread is causing this trouble). The main problem
with programming with threads is a controlled shutdown

JvanKatwijk, 2017-06-06:
It is no stupid question at all. What happens at start up is that - as soon
is there is time synchronization, i.e. the software "knows" where in the
samplestream the DAB frame starts - the DFT of block 0 - a block that
contains a characteristic pattern in the phases - is inspected to find that
pattern. Based on the offset of the located pattern in the DFT the coarse
offset is estimated and corrected. Since the pattern may not be clear,
there is a repeated search until the sofware is "certain" that the offset
id corrected. (The check is basically the ensemblename being recognized).

This coarse offset if completely handled in the thread handling the reading
and processing of the samplestream (i.e. the "ofdmProcessor").
Although it sounds attractive to apply the offset to the connecte device,
that is slightly problematic since there is a lot of buffering
between the device and the ofdmProcessor. Changing the frequency in the
device therefore is late.

It is obvious that if the oscillator in the device is correct (up to a few
Hz) the searching process for frequency synchronization is fast (depending
on the time needed to do the time synchronization).
Setting the offset in the device at startup - either through an offset in
khz or though a ppm offset therefore speeds up the process to do all
syncing needed for DAB recognition and processing.

As an example, I happen to have an Airspy device with a coarse offset of -
most of the times - 0, sometimes 1, and an SDRplay with a coarse offset of
-2. The airspy - which also has 12 bits rather than 8, synchronizes most of
the time within a second, the SDRPlay usually takes less than 2 seconds.
All done with an "antenna"  consisting of a piece of 40 cm wire, just
sitting in the livingroom.

I have two dabsticks, one pretty old, about 5 years, which requires a ppm
offset of app 84, the other one with another tunerchip requires an offset
of app 32.

So, summarizing: yes, in general when the ppm offset is correct,
synchronization is fast.

Hope this answers your question.

closed 2018-02-12 by JvanKatwijk