JvanKatwijk/qt-dab #31: "make install" not working

probonopd, 2017-03-25:
I can't get `make install` to work on Linux. Is it even supported?

JvanKatwijk, 2017-03-25:
you mean after "qmake-qt5" (or, depending on the distribtion/version)
"qmake" ?

probonopd, 2017-03-25:
Yes. Normally Qt applications can be compiled and installed using

```
qmake PREFIX=/usr
make
sudo make install
```

on Linux but that doesn't work here.

JvanKatwijk, 2017-03-25:
strange, it works for me. What is the output

probonopd, 2017-03-26:
Are you using cmake or qmake? It simply didn't do anything when I was using qmake.

probonopd, 2017-03-26:
This is what I get:

```
The command "make -j4" exited with 0.
0.14s$ sudo make DESTIDR=appdir install ; sudo chown -R $USER appdir ; find appdir/
[100%] Built target qt-dab-0.998
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/qt-dab-0.998
-- Removed runtime path from "/usr/qt-dab-0.998"
chown: cannot access ‘appdir’: No such file or directory
```

So it did not install stuff to `appdir/usr/bin/qt-dab`.

Here is the build script used:
https://github.com/probonopd/qt-dab/blob/cccfab06c271c132e6ff52dfaa4e949fe38917d6/.travis.yml

JvanKatwijk, 2017-03-26:
> 9fe38917d6/.travis.yml

To me it seems the qt-dab file was created, but installation failed.
sudo make DESTIDR=appdir install ; sudo chown -R $USER appdir ; find appdir/
are you sure this line is correct?

probonopd, 2017-03-26:
At least it works for many other applications...

JvanKatwijk, 2017-03-26:
It seems that DESTDIR /= DESTIDR

closed 2018-02-12 by JvanKatwijk

probonopd, 2018-02-12:
Thanks for catching and sorry for the noise. With `DESTDIR` it should hopefully work.