JvanKatwijk/qt-dab #32: desktop file and icon missing for Linux

probonopd, 2017-03-25:
[desktop file](https://standards.freedesktop.org/desktop-entry-spec/latest/) and icon are missing for Linux.

andimik, 2017-06-11:
You can create your own desktop file, see http://www.raspberry-projects.com/pi/pi-operating-systems/raspbian/gui/desktop-shortcuts

It should contain these lines:

```
[Desktop Entry]
Name=My Name
Comment=My application which does this
Icon=/usr/share/pixmaps/openbox.xpm
Exec=/usr/bin/leafpad
Type=Application
Encoding=UTF-8
Terminal=false
Categories=None;
```

probonopd, 2017-06-11:
Yes, but it would be nice if a "make install" would do this automatically. By the way, the `/usr/bin/` and `/usr/share/` can and should be left away.

closed 2018-02-12 by JvanKatwijk