JvanKatwijk/qt-dab #34: DAB stick not found with RTL2838

probonopd, 2017-03-26:
I get

> DAB stick not found! Please use one with RTL2832U or similar chipset!

I tried running Qt-DAB as normal user and as root (to exclude the possibility of missing permissions).

This is my device:

```
me@host:~$ lsusb
Bus 002 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
```

probonopd, 2017-03-26:
Hold on, may be related to a missing `librtlsdr.so` in the AppImage, which I will try to fix now.

probonopd, 2017-03-26:
Now I get 

```
Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
```

How do I enable automatic detaching at compile time?

probonopd, 2017-03-26:
Additionally, when running not as root, I see on the command line (but __not__ in the GUI)

```
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Opening dabstick failed
```

It would be nice if in this case the GUI would ask me to become root and fix it for me.

JvanKatwijk, 2017-03-26:
I experimented some time ago with an appImage, but one of the issues is
that the udev rules always should be installed,
so it is not "load and go: application and that is why I am not in favour
of i.
I do not think it is an issue for the application to solve

Wrt the kernel driver: that is another issue where the application does not
has the control. For some reason the library code of the rtl-sdr lib
contains a conditional (ifdef) to release the kernel driver or not.

That is why it is suggested to build your own librtlsdr.

probonopd, 2017-03-26:
Hi @JvanKatwijk thanks for the hint on compiling our own librtlsdr, I understand. However, for this to work, it would need to run as root, correct? Which would mean that the whole application would have to run as root?

probonopd, 2017-03-26:
@JvanKatwijk as for installing and loading `rtl-sdr.rules`, a small helper bash script, say `udev-rules-helper` could be included in the AppImage that would copy it into the system and load it. But the application is what controls the GUI, so if the application sees that the rules are not present, it should ask the user whether the helper script should be run, and if yes, execute `udev-rules-helper` with root rights.

JvanKatwijk, 2017-03-26:
No, "root" should install the librtlsdr, however, any normal user would be
able to run the application

probonopd, 2017-03-26:
How does librtlsdr then manage to unload a kernel driver without needing root permissions?

probonopd, 2017-03-26:
I seem to have a solution:

Instead of launching `qt-dab` directly, we launch this bash script:

```
#!/bin/bash
HERE="$(dirname "$(readlink -f "${0}")")"
pkexec "${HERE}/udev-rules-helper"
exec "${HERE}/usr/bin/qt-dab" "$@"
```

It invokes as root:
```
#!/bin/bash

HERE="$(dirname "$(readlink -f "${0}")")"

rmmod dvb_usb_rtl28xxu || true

cat > /tmp/10-rtl-sdr.rules <<\EOF
SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838", GROUP="adm", MODE="0666", SYMLINK+="rtl_sdr"
EOF

mv /tmp/10-rtl-sdr.rules /etc/udev/rules.d/10-rtl-sdr.rules

udevadm control --reload-rules
udevadm trigger --attr-match=subsystem=usb
```
and then finally runs `qt-dab`.

This seems to be working nicely for me on an Ubuntu Live ISO.

JvanKatwijk, 2017-03-26:
The part unloading the kernel driver is "ifdeffed", see the lines in
librtlsdr.c

jviotti, 2017-03-27:
Hi @probonopd 

> I seem to have a partial solution (to be refined):

Is there somewhere I could try this myself (e.g: a branch in `qt-dab`, or a patch you can share with me)?

> even though a mere "sudo su" can make me root there, and I do not have a user password

The way we handle this at Etcher is that first we try to run the command (e.g: `/tmp/udev-rules-helper`) using `sudo -n`. The `-n` means that `sudo` will behave non-interactively, and thus will not prompt for a password. After the program exits, we check `stderr` for `/sudo: /i`. If this matches, then it means that we need to prompt the user for elevation using something like `pkexec`, if not, then it means that the command actually did run with elevation already, so that's all there is to do. Check https://github.com/jorangreef/sudo-prompt/blob/master/index.js#L11 for details.

probonopd, 2017-03-27:
Hi @jviotti thanks for your hints, looks like this is really way more complicated that I expected.

Please see https://github.com/probonopd/qt-dab/tree/patch-1 and https://transfer.sh/SL994/Qt_DAB-git.fe13351-x86_64.AppImage for code and AppImage.

probonopd, 2017-03-31:
@jviotti when I remove my user from the sudoers file, then I get

```
me@host:~$ sudo -n -E -- ls
sudo: a password is required
```
So I do not get anything with `/sudo: /i`.

closed 2018-02-12 by JvanKatwijk