JvanKatwijk/qt-dab #36: slide show window

zygmund2000, 2017-03-30:
Hi,
Today I see slides show for first time, but new window always popup avery image change, this is very annoying.
Regards

andimik, 2017-03-31:
But if you want to see them again you find them in the same folder as the qt-dab executable

zygmund2000, 2017-03-31:
I kwow it but now each image open new window center on screen and covers others, eg center of web brower I see slides, I can't close it beceuse new image is again center of browser.

JvanKatwijk, 2017-03-31:
you want an option to prevent images to show up or do you want the
subsequent images as they show up to be on the same place?

andimik, 2017-03-31:
Why not opening only one popup window called slideshow where all the pics will be displayed? 

Same with the new epg function, in muxes where pics in the epg are transmitted.

zygmund2000, 2017-03-31:
I see at the moment each image closes and opens new window with some atribute on top of all others and center of my screen, good option will be always same window with manual position.
btw sometimes slides are displayed sometimes not, so maybe something is still wrong.

JvanKatwijk, 2017-03-31:
I just added a feature that once in a channel a picture is shown, the
picture label will remain the same.
Whether or not slides are shown largely depends on passing some crc checks
for the data. If the check fails, no image is shown.
I know there might still be something wrong, will be looking at it

zygmund2000, 2017-03-31:
After update I don't see slides and sound is unstable again :(

JvanKatwijk, 2017-03-31:
Strange everything works fine here, so I suggest you go back to the
previous version

zygmund2000, 2017-03-31:
check again and sound is ok but slides are not displayed.

andimik, 2017-03-31:
No, slides are displayed in my sdr or raw files I made from Bratislava, Vienna and Slovenia. 

Check console output, if you have this line:

`going to show picture img01.jpg`

then this window might be in the background.

For example: https://sdr.kt.agh.edu.pl/sdrdab-decoder/download.html , even in the 41 MB small file  antena-1_dab_229072kHz_fs2048kHz_gain42_1.raw there is a slideshow in Radio Kraków:

[image]

andimik, 2017-03-31:
Jan, this window should have a fixed size. Because if the previous pic was smaller the new one (max. 320x240) the new slide is cropped.

JvanKatwijk, 2017-04-01:
That is why I originally had a new window for each picture

andimik, 2017-04-01:
Sure, but the size is limited to 320x240, so you could open this window exactly in this size.

JvanKatwijk, 2017-04-01:
No, I do not think this is what you want

andimik, 2017-06-11:
The new versions don't show the slideshow in Windows (not tested in Linux so far)

> 2 samplerates are supported
> 6000000
> 3000000
> selected samplerate = 3000000
> Starting 1
> 
> we found for Slo 2 Val 202    serviceId 9202
> we found for Slo 2 Val 202    serviceId 9202
> we have now DAB+
> we found for Slo 1 Prvi       serviceId 9201
> we found for Slo 1 Prvi       serviceId 9201
> we have now DAB+
> cannot write file %tmp%\qt-pictures\rtvslo.jpg
> setGeometry: Unable to set geometry 5x13+363+124 on QWidgetWindow/'QLabelClassWindow'. Resulting geometry:  120x13+363+124 (frame: 8, 31, 8, 8, custom margin: 0, 0, 0, 0, minimum size: 0x0, maximum size: 16777215x16777215).

closed 2018-02-12 by JvanKatwijk