JvanKatwijk/qt-dab #37: Text missing in file open/save window

andimik, 2017-03-31:
Seems the texts in brackets in the "file open" or "file save" window are not displayed. See radio.cpp, lines 497, 1218, 1233, 1248, 1461 and 1496.

For example:

` tr ("iq data (*.iq)"));`

I also tried escaping, but then there is a compiler warning ...

> warning: unknown escape sequence: '\)'

JvanKatwijk, 2017-04-01:
Not sure what you mean. The text appears in the bottom right corner and the
search filter is limited to either "raw", "iq" or "sdr"

andimik, 2017-04-01:
Yes, I know where it is. But the text in brackets is missing. It only displays raw file instead of raw file (*.raw)

JvanKatwijk, 2017-04-01:
As far as I know the wildcard dot suffix (e.g. *.iq) is just used as filter
and not interpreted as part of the text,  see e.g.
http://stackoverflow.com/questions/1604440/how-to-set-selected-filter-on-qfiledialog

andimik, 2017-04-01:
You are right. I thought the filter was made 4 lines before ...

closed 2017-04-01 by andimik