JvanKatwijk/qt-dab #39: No longer compiles on RPI3

NomisGninnac, 2017-04-01:
Latest release refuses to compile, previous releases were able to compile.

------------------
 -Iincludes/scopes-qwt6 -Ioptional-scope -Iinput/dabstick -Iinput/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o rtl_tcp_client.o input/rtl_tcp/rtl_tcp_client.cpp
g++ -c -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -DQT_WEBKIT -DTECHNICAL_DATA -DMOT_DATA -DMSC_DATA__ -D__QUALITY -DHAVE_SPECTRUM -DHAVE_DABSTICK -DHAVE_RTL_TCP -DTRY_EPG -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Iinput -Iinput/rawfiles -Iinput/wavfiles -I/usr/local/include -I/usr/local/include -I/usr/include/qt4/qwt -I/usr/include/qt5/qwt -I/usr/include/qt4/qwt -I/usr/include/qwt -Iincludes/scopes-qwt6 -Iincludes/scopes-qwt6 -Ioptional-scope -Iinput/dabstick -Iinput/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o epgdec.o src/backend/data/epg/epgdec.cpp
src/backend/data/epg/epgdec.cpp: In member function ‘void CEPGDecoder::decode(const std::vector<unsigned char>&, const QString&)’:
src/backend/data/epg/epgdec.cpp:46:59: error: ‘fopen’ was not declared in this scope
     FILE *epgFile = fopen (name. toLatin1 (). data (), "w");
                                                           ^
src/backend/data/epg/epgdec.cpp:48:17: error: ‘stderr’ was not declared in this scope
        fprintf (stderr, "filename = %s\n", name. toLatin1 (). data ());
                 ^
src/backend/data/epg/epgdec.cpp:48:70: error: ‘fprintf’ was not declared in this scope
        fprintf (stderr, "filename = %s\n", name. toLatin1 (). data ());
                                                                      ^
src/backend/data/epg/epgdec.cpp:50:23: error: ‘fclose’ was not declared in this scope
        fclose (epgFile);
                       ^
src/backend/data/epg/epgdec.cpp:53:17: error: ‘stderr’ was not declared in this scope
        fprintf (stderr, "cannot open %s\n", name. toLatin1 (). data ());
                 ^
src/backend/data/epg/epgdec.cpp:53:71: error: ‘fprintf’ was not declared in this scope
        fprintf (stderr, "cannot open %s\n", name. toLatin1 (). data ());
                                                                       ^
src/backend/data/epg/epgdec.cpp: In function ‘QDomElement element(QDomDocument&, const tag_length_value&)’:
src/backend/data/epg/epgdec.cpp:332:27: warning: comparison is always false due to limited range of data type [-Wtype-limits]
 #define invalidTag(x) ((x < 0) || (x > 50))
                           ^
src/backend/data/epg/epgdec.cpp:335:6: note: in expansion of macro ‘invalidTag’
  if (invalidTag (tlv. tag)) {
      ^
Makefile:1171: recipe for target 'epgdec.o' failed
make: *** [epgdec.o] Error 1

andimik, 2017-04-01:
And when you comment out the line

    CONFIG	+= try-epg

In my case this function does not create a textfile as described in the readme, but some small icons.

JvanKatwijk, 2017-04-01:
You should not enable try-epg, that is really "work in (slow) progress

NomisGninnac, 2017-04-01:
Code now compiles. as of update.  Possibly error in my editing of .pro file.  Please feel free to close this issue.

Still dont get dynamic label displaying.  I'm unable to work this out. But its not a massive problem without it.

closed 2017-04-01 by JvanKatwijk