JvanKatwijk/qt-dab #40: missing file

zygmund2000, 2017-04-10:
I can't compile from git:

CMake Error at CMakeLists.txt:506 (add_executable):
  Cannot find source file:

    ./input/virtual-input.cpp

Regards

JvanKatwijk, 2017-04-10:
Oops, I was renaming the directory input into devices (just to bring things
in line with other work), done

zygmund2000, 2017-04-10:
So now is another error:

CMake Error at CMakeLists.txt:508 (add_executable):
  Cannot find source file:

    ./devices/airspy/airspy-handler.cpp

JvanKatwijk, 2017-04-10:
You are absolutely right, "airspy", the directory was changed in
airspy-handler.
This is one of the problems with github (and comparable systems), the
latest is always "work in progress"
(It is changed btw)

zygmund2000, 2017-04-10:
Can you check evertyhing? because another error

CMake Error at CMakeLists.txt:508 (add_executable):
  Cannot find source file:

    ./devices/dabstick/dabstick.cpp

JvanKatwijk, 2017-04-10:
Sorry, now it compiles

zygmund2000, 2017-04-10:
but now after compilation dabstick is not available?

JvanKatwijk, 2017-04-10:
set (${objectName}_SRCS
                ${${objectName}_SRCS}
                ./devices/rtlsdr-handler/rtlsdr-handler.cpp
                ./devices/rtlsdr-handler/rtl-dongleselect.cpp
           )

           add_definitions (-DHAVE_RTLSDR)
        endif()
#

Later this week I'll make a kind of stable release, I am currently making
quite some changes and cannot garantee that every evening all tests are
performed. For now: if you change the CMakeLists.txt file such that
HAVE_DABSTICK is replaced by HAVE_RTLSDR  then it should work for the stick

zygmund2000, 2017-04-10:
Ok, I'll wait for it.

closed 2017-07-10 by JvanKatwijk