JvanKatwijk/qt-dab #41: Compiling on RPi3

NomisGninnac, 2017-04-11:
Getting this error on make:

pi@raspberrypi:~/qt-dab $ make
g++ -c -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -DQT_WEBKIT -DTECHNICAL_DATA -DMOT_DATA -DMSC_DATA__ -D__QUALITY -DHAVE_SPECTRUM -DHAVE_RTLSDR -DHAVE_RTL_TCP -DTII_ATTEMPT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Idevices -Idevices/rawfiles -Idevices/wavfiles -I/usr/local/include -I/usr/local/include -I/usr/include/qt4/qwt -I/usr/include/qt4/qwt -I/usr/include/qwt -Iincludes/scopes-qwt6 -Iincludes/scopes-qwt6 -Ioptional-scope -Idevices/rtlsdr-handler -Idevices/rtl_tcp -I../janSpul -I. -I. -o main.o main.cpp
In file included from radio.h:40:0,
                 from main.cpp:33:
includes/ofdm/ofdm-processor.h:38:26: fatal error: tii_detector.h: No such file or directory
 #include "tii_detector.h"
                          ^
compilation terminated.
Makefile:583: recipe for target 'main.o' failed
make: *** [main.o] Error 1

JvanKatwijk, 2017-04-11:
#if you uncomment this one, you will get lots of compiler errors, it is
just private
 CONFIG += try_tii # sorry, just private stuff

This line is - for whatever reason - in the .pro file uncommented, just
comment it out

andimik, 2017-04-11:
Jan, maybe some of the problems from the last days could have been reduced by working with a stable branch "master" and test new functions in a separate branch.

JvanKatwijk, 2017-04-11:
Yes, but this comment does not make much sense, since the pro file does
contain a clear statement to comment this line out. There were some issues
in the CMake file due to applying a different naming convention (due to
changes in a local -non github- project -, but as far as I am concerned,
the current qt-dab version is as stable as it can be. I plan to generate a
new windows version for it and create a "release"

andimik, 2017-04-11:
You are right. It is stated very clearly! One should realize it as the mentioned directory is local, not part of the master branch.

NomisGninnac, 2017-04-11:
Many thanks for the update.  I did try to find the relevant include to comment out, but alas was a bit blind perhaps.

I have successfully compiled todays edition OK.  Many thanks.

JvanKatwijk, 2017-04-12:
Good to hear. I have no intention to further change the current version for
some time since I am working on something different
best
jan

closed 2017-07-10 by JvanKatwijk