JvanKatwijk/qt-dab #42: Sound

zygmund2000, 2017-04-15:
Actulal version from git has sound problems again, I have also old sdr-j .9985 and sound is ok.

JvanKatwijk, 2017-04-15:
Can you be more specific

zygmund2000, 2017-04-15:
I can, but what do you want to know? sound drops after update from git

JvanKatwijk, 2017-04-15:
Well, since nothing changed in the handling of sound since the discussions
we had some time ago, I need more information - or samples - to investigate
what is going on.

zygmund2000, 2017-04-15:
I restarted my system and sound is still not stable, drops ocuurs randomly, maybe It's not related to sound handling but something else, I see also changes in fft view refresh (btw old view was better).

JvanKatwijk, 2017-04-15:
Compared to what? nothing changed in the fft handling nor viewing since
Qt-DAB started, so are you comparing with sdr-j-dab or with earlier
versions of Qt-DAB?
I do not mind getting comments, but pls give me something that I can use to
address the issue

zygmund2000, 2017-04-15:
Compared to version 1-2weeks ago, today fft is different like less averaged, I don't have screens from past, I really don't kwow what strange things are going now, before last compilation averything was ok

zygmund2000, 2017-04-15:
I found this old workng qt-dab on other pc and copied to this and works all ok, so I'm really cant't help you

zygmund2000, 2017-04-15:
maybe it's problem with my new compilation, meybe newer libraries since 1-2 weeks

zygmund2000, 2017-04-15:
No, it's not , I cleaned and compiled old code again and everything is ok.
After it I compiled now new code from git again and still not good.

zygmund2000, 2017-04-15:
http://i.imgur.com/aVXgM17.png - old fft view
http://i.imgur.com/cLg2aSM.png - new fft view

closed 2018-02-12 by JvanKatwijk