JvanKatwijk/qt-dab #44: CMAKE_INSTALL_PREFIX not honoured

russel, 2017-05-01:
I use the CMake build and set -DCMAKE_INSTALL_PREFIX for the build area. However install is going to /usr/local/bin even though this override destination is set.

JvanKatwijk, 2017-05-01:
The CMakeLists.txt file is adapted such that the default case is /usr/local/bin, which can be overrridden by setting -DCMAKE_INSTALL_PREFIX= YYYYYY

closed 2017-05-01 by JvanKatwijk

russel, 2017-05-01:
But that is what I am saying, the option is not working. I set -DCMAKE_INSTALL_PREFIX=$HOME/Built and this in not honoured, I still get install to /usr/local/bin.

JvanKatwijk, 2017-05-01:
I changed the CMakeLists.txt file and I can install it whereever I like to

russel, 2017-05-01:
My apologies, I replied before I updated correctly. The fix is in and works. Thanks. Again apologies for reacting too quickly in words.

JvanKatwijk, 2017-05-01:
No problem, your comment forced me to read into cmake stuff, something that
was on my todo list, but continuously shifted backwards