JvanKatwijk/qt-dab #47: compilation error

zygmund2000, 2017-05-28:
I have compilation error:

[100%] Linking CXX executable qt-dab-0.999
/tmp/ccYqk3rV.ltrans5.ltrans.o: In function `mp4Processor::~mp4Processor()':
<artificial>:(.text+0x123): undefined reference to `faadDecoder::~faadDecoder()'
/tmp/ccYqk3rV.ltrans5.ltrans.o: In function `mp4Processor::~mp4Processor()':
<artificial>:(.text+0x2953): undefined reference to `faadDecoder::~faadDecoder()'
/tmp/ccYqk3rV.ltrans5.ltrans.o: In function `mp4Processor::handle_aacFrame(unsigned char*, short, unsigned char, unsigned char, unsigned char, unsigned char, bool*)':
<artificial>:(.text+0x45ba): undefined reference to `faadDecoder::MP42PCM(unsigned char, unsigned char, short, unsigned char, unsigned char*, short)'
/tmp/ccYqk3rV.ltrans11.ltrans.o: In function `mp4Processor::mp4Processor(RadioInterface*, short, RingBuffer<short>*)':
<artificial>:(.text+0x1a1e): undefined reference to `faadDecoder::faadDecoder(RadioInterface*, RingBuffer<short>*)'
<artificial>:(.text+0x1bb7): undefined reference to `faadDecoder::~faadDecoder()'
/tmp/ccYqk3rV.ltrans20.ltrans.o:(.data.rel.ro+0x3c8): undefined reference to `faadDecoder::~faadDecoder()'
/tmp/ccYqk3rV.ltrans20.ltrans.o:(.data.rel.ro+0x3d0): undefined reference to `faadDecoder::~faadDecoder()'
collect2: error: ld returned 1 exit status

I hope it's helpful.

zygmund2000, 2017-05-30:
Now is ok but SNR level is always 0?

JvanKatwijk, 2017-05-30:
Interesting error, it turned out that a C comment (/* .. */) was not
closed, such that a few lines of code were commented out

zygmund2000, 2017-05-31:
BTW: SNR calculation seems to be lower than it should, now maximum is about 22 but real value could be up to 33.
What do you think about it?

JvanKatwijk, 2017-05-31:
The value shown is not exact. What is done is that the power of a segment
within the spectrum of the signal is compared with a segment
outside the spectrum of the signal. If - as is the case with the SDRplay -
there is filtering of the signal, then the value of the segment
outside the spectrum will be less than in case there is not filtering. For
a dabstick like device, there is an 8 bit signal, i.e. a max of 48 db
(theoretically), so you have to have a pretty good signal for a 20 - 25 db
SNR. But again, it is an indication, no more no less.

zygmund2000, 2017-05-31:
I have very good signal, if I look at spectrum in eg gqrx with dabstick difference between top and near background noise is almost 38dB.
Many many versions ago SDR-J had better constellation view and SNR range to 34dB.

JvanKatwijk, 2017-05-31:
I'll look into it. Can you send me a sample of a really good signal?

zygmund2000, 2017-06-01:
I sent you some time ago sample file but when I play it looks different than live from dabstick.

JvanKatwijk, 2017-06-01:
do you know the name of the file? I have many files here

In the mean time I did some measurements on the signal from my dabstick
signal with antenna connected, however tuned to a channel without DAB
ensemble, average signal value 0.063052
signal without antenna connected
                                                            0.010587
signal strength of the NULL period (i.e. no transmitted value)  when tuned
to a decent ensemble              0.030664
signal strength of Block 0 when tuned to the same channel
                                                0.110449

the indication we get with the last one is 11 db. 20 log (0.1105/0,0306) is
app 11.

Note that the snr as displayed is computed from the values in the spectrum.
It might be better to compute the snr from the ratio
average signal value block 0 / average signal value null period

JvanKatwijk, 2017-06-01:
I ran some interesting tests. First question is what do you want the db
indicator to indicate?
The current setup gives the average power of the signal within the
spectrum, which obviously strongly relates to the
reference value, i.e. the noise level you take into account. The noise
level taken is the average power of the part of the spectrum not belonging
to the DAB signal. The fft is taken from Block 0.

The other setup gives the power of the signal in the time domin, i.e. the
amplitude of the samples, with as reference value the
average signal value in the null period of the DAB frame, i.e. effectively
the noise level in the incoming signal.

The measured values - in setup 2 - are smaller than the ones in setup 1.
The interesting thing is that when I change the gain
in the device (I am using a airspy right now), the setup 1 reports lower
snr, while the setup 2 is the level level.

2017-06-01 12:54 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> do you know the name of the file? I have many files here
>
> In the mean time I did some measurements on the signal from my dabstick
> signal with antenna connected, however tuned to a channel without DAB
> ensemble, average signal value 0.063052
> signal without antenna connected
>                                                               0.010587
> signal strength of the NULL period (i.e. no transmitted value)  when tuned
> to a decent ensemble              0.030664
> signal strength of Block 0 when tuned to the same channel
>                                                   0.110449
>
> the indication we get with the last one is 11 db. 20 log (0.1105/0,0306)
> is app 11.
>
> Note that the snr as displayed is computed from the values in the
> spectrum. It might be better to compute the snr from the ratio
> average signal value block 0 / average signal value null period
>
>
>
>> I sent you some time ago sample file but when I play it looks different
>> than live from dabstick.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

zygmund2000, 2017-06-01:
I also think that is very difficult to calculate real SNR value, teoretical 'SNR' for dab ofdm signal is about 33dB because carriers cover other carriers about this point. But I think range of SNR calculation should be up to 33dB, more is not possible even with if higer dynamic receiver. Maybe calculation SNR from constellation?

JvanKatwijk, 2017-06-01:
When running the file 10d_lang2 - someone pointed to the link, you may be?
- gives an SNR of 33, that is indeed an excellent signal.

zygmund2000, 2017-06-01:
I sent you file with test1, test2 etc.. stations.

JvanKatwijk, 2017-06-01:
That file was probably deleted when cleaning up my computer (I has well
over 100 G of dab files).
However, with a file downloaded from a polish site we get an snr of 32-34
so, yes it is certainly possible to get reasonable snr values

zygmund2000, 2017-06-02:
I see some changes in SNR now I have 39-41 it looks a bit too high

JvanKatwijk, 2017-06-02:
OK,
as you know, the snr is computed as 20 * log10 (signallevel / noiselevel).
In order to prevent one of the two values from zero I had added a value
1.0, but since the chance that these values are zero is small, I reduced
these additions. And as you know x / y > (x + 1) / (y + 1)

The computation is done in the file ofdm-decoder.cpp, if you look for the
pattern "log10", you can modify the addition.

zygmund2000, 2017-06-02:
I know principles of SNR but how far do you take noise to calculation form main signal?

JvanKatwijk, 2017-06-02:
The part of the spectrum not belonging to the DAB signal is considered
noise. So, it is computed by looking at a segment of the spectrum in the
dft of the Block 0 of the DAB frame. We know the carriers where the signal
is, so just compute the average of the amplitude values
of the spectrum within and outside the DAB signal (The spectrum has 2048
carriers, where 1536 carriers belong to the DAB signal
and the remaining ones not. The amplitude of these remaining ones are taken
as measure for noise.)
It is a pretty simple computation, the only difference with previous
versions in the code is now that the additions are much lower.

zygmund2000, 2017-06-02:
Yes but my question was how far you measure noise from end of -768/+768 carrier? Because noise at eg 790 will be too high ang give error in calculations.

JvanKatwijk, 2017-06-02:
The noise is computed over the carriers + 998 .. 1024 + -1024 .. -998, so
the far end at both sides

JvanKatwijk, 2017-06-02:
oeps 924 .. 1024 en -1024 .. - 924 (200 carriers are taken)

2017-06-02 20:51 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> The noise is computed over the carriers + 998 .. 1024 + -1024 .. -998, so
> the far end at both sides
>
>
>
>> Yes but my question was how far you measure noise from end of -768/+768
>> carrier? Because noise at eg 790 will be too high ang give error in
>> calculations.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

zygmund2000, 2017-06-02:
Now is clear but I'm afraid about this method because if you have strong multipath signal and holes within ofdm... But I don't have any better idea at this moment.

JvanKatwijk, 2017-06-02:
That is why it needs to pass a moving average or other filter.

closed 2017-06-04 by JvanKatwijk