JvanKatwijk/qt-dab #48: Path of Image Files

HawkingJan, 2017-05-31:
Hi,

I noticed that there are several image files (like  slide11.jpg  slide22.jpg  slide91.jpg  slide94.png ). Most of them are in the same directory like were the binaries are located (linux-bin folder). However I also found one image in my home directory (most likely because I started qt-dab from there using  `~/Programms/qt-dab/linux-bin/qt-dab-0.999` ). I don't know if these files are leftovers for cases were I had to kill the programm (e.g. because I lost the connection to my Raspberry). Anyway I would strongly prefer if these (temporary) images would be placed in some temporary folder (like /tmp/qt-dab ) and not all-over the place.

andimik, 2017-05-31:
BTW Images are only stored when the program detects a filename, otherwise they are just displayed. But each slide has indeed a filename according to the eti content.

JvanKatwijk, 2017-05-31:
Seems a fair suggestion to have a predefined location for storing pictures.
My suggestion then would be to have a default location somewhere in /tmp
(does windows have a /tmp?) and a possibility to set the location in the
ini file.

JvanKatwijk, 2017-05-31:
A first attempt is now in the git. "pictures" in the ini file will
define/set a location to store the files and some other things that are
saved.
Default /tmp/qt-pictures

2017-05-31 15:42 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Seems a fair suggestion to have a predefined location for storing
> pictures. My suggestion then would be to have a default location somewhere
> in /tmp (does windows have a /tmp?) and a possibility to set the location
> in the ini file.
>
>
>
>> Hi,
>>
>> I noticed that there are several image files (like slide11.jpg
>> slide22.jpg slide91.jpg slide94.png ). Most of them are in the same
>> directory like were the binaries are located (linux-bin folder). However I
>> also found one image in my home directory (most likely because I started
>> qt-dab from there using ~/Programms/qt-dab/linux-bin/qt-dab-0.999 ). I
>> don't know if these files are leftovers for cases were I had to kill the
>> programm (e.g. because I lost the connection to my Raspberry). Anyway I
>> would strongly prefer if these (temporary) images would be placed in some
>> temporary folder (like /tmp/qt-dab ) and not all-over the place.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

HawkingJan, 2017-05-31:
Thanks for this quick change (I didn't test it yet but it sounds OK). Just three follow up questions that came to my mind:
1) Is there a switch to disable the display of the images (set during compilation or as preferred variant via the ini file) ?
2) If I understood andimiks comment correct the images don't need to be saved to be displayed. Is there a way to show the images but prevent them from being saved at all (I might be a bit over concerned but the space on my /tmp directory is limited and even if the images are small they might occupy more and more space while the qt-dab is running). A automatic delete job would be dangerous (if the user sets this variable by accident to his home...) but in my naive world preventing the image from being saved at all by setting a flag in the ini-file should be easy
if (image with filename detected and saveimageflag == 1) then saveimage
3) You said that the path is also used to save other information. Just out of interest, what type of information and could this data also fill the path during normal run time? 

These are not hard issues, so please let me know if you want me to create extra "request tickets" for them.

JvanKatwijk, 2017-05-31:
ad1
set saveSlides=0 in the ini file
ad 2
There is currently no switch that controls the saving of the pictures. The
most simple solution might be to refuse saving pictures if the
path is set to "" (Look into that tomorrow)
ad 3
If/when you compile with the MSC_DATA on and select a program marked data
and it is MOT data then files with data will be saved (sometimes
directories with html/xml type data)
The problem in developing is that her ein the Netherlands no data is
transmitted, so development is pretty slow. However from time to
time I do some development on epg and tpg data.

HawkingJan, 2017-06-02:
saveSlides=0 to disable the display of images (that don't need to saved to be displayed) is not really intuitive but thanks.

I remembered MSC_DATA and MOT_DATA as separate options from the compilation. Do I understand your post correct that it makes no sense to compile with MOT but without MSC?

JvanKatwijk, 2017-06-03:
That is true. The MSC option, when switched off, causes the data channels
not to be visible to the user. The MOT option
prevent some interpretation from MOT data in the data channels.

afaik saveSlides relates to saving the picture while showPicture (better
name would have been showSlides) controls showing the
images that are contained in the audio programmes

HawkingJan, 2017-06-03:
> saveSlides relates to saving the picture 

You mentioned saveSlides as answer to my question related to the "display of the images". That's why I got a little confused. 

Maybe you can add short paragraph about saveSlides, showPicture and MOT_DATA at https://github.com/JvanKatwijk/qt-dab and the corresponding part of the config or ini file.

From my point of view this issue can be marked as closed.

closed 2017-06-04 by JvanKatwijk