JvanKatwijk/qt-dab #50: Stop/Pause Playback?

HawkingJan, 2017-06-05:
Maybe I'm missing an obvious solution, but is there a way to stop the playback of music without closing the complete program (I assume that setting the gain in the extra windows to 0 would still use resources?)
Instead of adding an extra button one could maybe think about toggling the "start" button to a "stop" button while music is being played?
(Just as a side question, what is the difference between the "Start" and the "Scan band" button?)

JvanKatwijk, 2017-06-06:
There is no "pause" button. Setting the gain does not stop the process -
apart from the fact that synchronization is lost.
In (very) early versions there was a pause button, but that went lost when
simplifying the GUI.

Adding a feature for pausing when replaying a file is not very complex.
When I am back I'll consider that.

The difference between the start and the scanbutton is that the start
starts the program regardless what the other parameters are, while the scan
band button starts a scanning process, stopping as soon as it believes
there is an ensemble in the then selected channel.

andimik, 2017-06-06:
There are some ways as a workaround :

Change to an unused VHF channel 
If applicable select an EPG or other data service

closed 2018-02-12 by JvanKatwijk