JvanKatwijk/qt-dab #52: RTL-TCP port fixed to 1234...

lars18th, 2017-06-27:
Hi,

At time the RTL-TCP port is fixed to 1234... however this port can be in use in the server.

It's possible to support an user configurable port?

Thank you!

JvanKatwijk, 2017-07-05:
I was on holidays in June so (Internet free, can recommend it), so I jst
saw your mail.
I'll look into it next week when I am more or less settles again

best
jan

JvanKatwijk, 2017-07-07:
Hi
I looked into the code and I have seen that if you set rtl_tcp_port in the
ini file to the port you want, that port is used.
best
jan

lars18th, 2017-07-10:
Hi @JvanKatwijk ,

> I looked into the code and I have seen that if you set rtl_tcp_port in the
ini file to the port you want, that port is used.

Great!
Thank you.

closed 2017-07-10 by JvanKatwijk

quartapound, 2019-02-05:
> Hi I looked into the code and I have seen that if you set rtl_tcp_port in the ini file to the port you want, that port is used. best jan 2017-06-27 21:10 GMT+02:00 lars18th <notifications@github.com>:
> […](#)
> -- Jan van Katwijk +31 (0)15 3698980 +31 (0) 628260355

Hello! I'm trying to do the same thing (change the default port from 1234..... multipsk only allows ports above 5000.  where does this ini file reside!?  I've been googling for awhile without much success...

tried starting rtl_TCP in rc.local with :5005 added to the end and it doesnt work....  everything works great with the default port, how the heck do you change it? (and WHERE!?)

Thanks!

lars18th, 2019-02-05:
> tried starting rtl_TCP in rc.local with :5005 added to the end and it doesnt work.... everything works great with the default port, how the heck do you change it? (and WHERE!?)

Read the section "Comment on some settings" in the initial README... It says:
> Some values of settings are maintained between program invocations. This is done in the (hidden) file .qt-dab.ini in the user's home directory.

This is the code line that sets the value:
https://github.com/JvanKatwijk/qt-dab/blob/14a456df4196b4409c25f70f8441eff6db33be7c/devices/rtl_tcp/rtl_tcp_client.cpp#L52
So, it's "1234" as default if no INI file.

Regards.

JvanKatwijk, 2019-02-05:
If the "ini" file contains a pair "rtl_tcp_port=abcd", where abcd is the
portnumber of your favorite port then the software will
(try to) take that rather than the default "1234"