JvanKatwijk/qt-dab #53: Protection level +1 in console window

andimik, 2017-07-12:
In fib-processor.cpp I found protlevel 0, 1, 2, and 3, but the best protection level is 1A, not 0A.

That's why an EPG or TPEG (Slovenia, Vienna, Italy) shows 

> data: protlevel = 2

instead of 3.

So you should add 1 for the output.

Hint: The output in the "Technical Data" window is - on the other hand - correct for all audio services.

JvanKatwijk, 2017-07-12:
Hi Andreas

Good to hear from you, it has been a while.

The encoding of the protection levels in the FIB blocks is 0 .. 3, can you
be more specific where/how EPG/TPEG show the prot leve

best
jan

andimik, 2017-07-12:
[image]
Sure, I wanted you to have a good vacation in the meanwhile ;-)

Well, I know, that the protection level for the EPG in Slovenia (called "Spored") is 3A (see http://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=SVN&inputEn=9200 as well), but the console output says 

> data: protlevel = 2

Whereas for audio services the information in the window is correct (3A in this case)

[image]

JvanKatwijk, 2017-07-12:
> inputLand=SVN&inputEn=9200 as well), but the console output says

I had a quick look into the code: what is printed is the protection level
as passed on from the FIB structure. When handled
it will act as the protLevel + 1. Rhat you see the correct level in the
technical data is because the internal number (i.e. the one
in the range 0 .. 3) is translated

best
jan

closed 2018-02-12 by JvanKatwijk