JvanKatwijk/qt-dab #54: 2nd open file dialog

andimik, 2017-07-30:
I'm using Qt-DAB mainly for prerecorded files. 

But it seems (obviously since some versions) that when I chose a raw/iq/sdr file at the start the same window appears again and I have again to specify the file name.

JvanKatwijk, 2017-08-01:
Not completely true. It happens when you click (twice) on the filename. If
you click on "open", then the file
will open directly.

I noticed this strange behaviour but interaction with a filedialog is a
library function, provided for by Qt.

andimik, 2017-08-01:
Ok, Thanks for the hint. I'll try it and close this issue.

andimik, 2017-08-10:
Don't know if this helps

https://forum.qt.io/topic/41356/qfiledialog-signal-fileselected-is-triggered-twice-for-one-user-choice-changed-in-recent-qt
https://riverbankcomputing.com/pipermail/pyqt/2010-February/025869.html

andimik, 2017-08-13:
Seems this was fixed by chance ...

Does not appear :-)

andimik, 2017-10-01:
With 1.0 alpha I have much more problems, the open file dialog now opens up to 8 times. This is very annoying and should be fixed asap.

I checked older versions from February, this does NOT appear.

JvanKatwijk, 2017-10-02:
I am sorry to hear that. I was just happy that on my Fedora box and under
windows that problem seemed to be gone.
What system you are running?

ps
there are some issues left though, one of them is handling the rtl_tcp

andimik, 2017-10-19:
Let's close it. Seems really to be solved. :-)

closed 2017-10-19 by andimik