JvanKatwijk/qt-dab #55: Raspberry Pi 3 Dropouts

sjdean, 2017-07-31:
DAB on the raspberry pi 3, is intermittent. I've had anywhere between 1 minute and 15 minutes audio before it just grinds to a halt. It appears that the CPU remains high at 60% yet no audio.

If you can direct me how to test and provide feedback, I will. No errors that I've seen. No crashes. Just a stop in audio.

I'm using a Nooelec NESDR SMART device. Not entirely convinced by it. But reviews say it's just as good as the RTL SDR dongle from rtlsdr.

Thanks
Simon

NomisGninnac, 2017-08-01:
I have a similar problem on my RPi3.  

When signal is weak, the channel list blanks and there is a loss of connection to the USB dabstick.

Reselecting dabstick and hitting start gets it all working again.

JvanKatwijk, 2017-08-02:
Oops, Difficult to say something useful from here. I have an RPI 2 and just
- as a test - had the software running for about two hours,
The audio is obviously depending on the quality of the incoming signal.
What does the "technical data" say?
The four indicators at the bottom all relate to the incoming data. If these
remain 100% with intermittent audio output then
something is wrong with the final step: handling the audio

In most cases that I have intermittent sound, the second indicator is not
up to 100%, that indicator tells how many DAB fields are
correct.

So, the first step in detemining what is going wrong is to detemine whether
the signal is weak or the output is wrong

A simple test is to send the audio out to a file. If the sound there is OK
it is the audio out, else the data in
j

Phil1951, 2017-08-17:
I have RPI 3B, Newskey RTL2832U, running standard Raspbian. I'm using Qt4. I needed to install libsamplerate and libqwt, then edit the -L declarations on the Makefile, but after that, it compiled OK. It seems to run OK with between 27% and 40% core usage on the RPI 3B. Is the original poster trying to do other work on the RPI when Qt-DAB is running ? When setting up for the compile, I found it a great help to check libraries etc using "Synaptic Package Manager". But yes. it will work OK on the RPI 3B. Thanks, Phil Dodd

JvanKatwijk, 2017-08-18:
Thanks for the info. On the RPI2 cpu usage is app 60 - 70 percent. I still
have to buy an RPI3
best
jan

andimik, 2017-08-18:
I will try in two weeks on my Rpi 3. But I am using only file input. And spectrum is not configured as I don't know how it works in qt4.

JvanKatwijk, 2017-08-18:
Well, for the spectrum you need qwt, and raspbian (at least on the rpi2)
supports qt5 without and qt4 with qwt.

Phil1951, 2017-08-18:
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

Jan,

Thank you for your reply.

I wanted to make sure that experimenters were not left with an impression
that there was some problem with the RPI 3B. Two experimenters had reported
problems. It was important that I stated that my own installation is
working perfectly. I suspect that the first experimenter has a DABSTICK
which is not suitable for the project, or for which the driver is not
correct. He also failed to say which distribution that he was using.

Your recent rationalisation, to create a single set of programs has been a
good success - congratulations to you upon making that big change, which
must have been a large amount of work for you.

The RPI3 is indeed a good processor board. The only aspect of the RPI that
I do not enjoy is the way that SD cards are inserted and removed into the
card reader. With your project, though, this should not be a problem, as
the same SD card will remain in use for all of the time.

I wanted to replace a closed-source DAB radio with an open-source one. I
bought a DAB radio in the UK, only to find that the manufacturer was unable
to update the firmware to decode DAB+ , despite the firmware being in use
in Australia on the same radio. Your excellent project has enabled me to do
that. In the UK, we still only have a few DAB+ channels. I do not think
that the BBC will change to DAB+ very soon, because so many DAB radios are
already in use here without being able to decode DAB+. In the UK, the
implementation of DAB has been a very unsatisfactory exercise. The BBC
started with a few regional DAB transmitters, 15 years ago. They now have
over 300 transmitters across the UK, to provide enough signal for everyone
to receive a strong signal !

Your Qt-DAB software is excellent ! Congratulations ! Thank you !  I will
continue to follow your project !

Good luck with ordering the RPI 3 !

Philip Dodd
Stoke Prior
LEOMINSTER
Herefordshire
UK
( half-way between Birmingham and Cardiff )

Phil1951, 2017-08-21:
Sorry about so many comments - re RPI, on 16 August, Raspbian has been updated to Debian Stretch, which uses Qt5. The files that need to be installed for Qt-DAB therefore need to be Qt5 ones from this release. The Qt5 alternatives are mostly obvious. There is one name change, in that libqt4-opengl-dev is replaced by libqt5opengl5-dev .

As a novice programmer with Qt, I found it essential to use Synaptic Package Manager for the installation as per the README.md file, and "mc" as a backup check.

I spent a long time trying to get qmake to work, which generated a Makefile with an -I list that needed a good deal of tidying, and main.cpp couldn't resolve "include QAPPLICATION".

I therefore installed and tried cmake, copied the Qt5 "txt" file to my compiling "txt" file, and it compiled with no major problems... there are a good number of warnings relating to unused variables...

I decided to use the "dabstick" RTL ( and its development library ) that is already available on the repositry, which works OK.

Regarding running Qt-DAB, I have found that the gain is best set to around midway - on mine at 19 rather than 42 - and it seems OK... 

So I recommend that anyone installing Raspian Stretch, compiles Qt-DAB using "cmake"...

Regards, Phil Dodd

JvanKatwijk, 2017-08-21:
Thanks

I'll try the newest Raspbian soon. One of the problems with Linux is that
different distributions always have things differently.
In order to make an appimage for X64 PC's, I had to write a CMakeLists.txt
file for Qt4, and that seems to work.

The issue with Qt5 on the RPI was not that there was not a Qt5 library, the
qwt library , used for showing the spectrum - if so configured -
was compiled against Qt4, and (afaik) not available for Qt5. While it is
certainly possible to compile qwt for Qt5, it is not a trivial process.

best
jan

Phil1951, 2017-08-21:
Thanks for your reply and explanation.

I have run some checks on the memory usage of the RPI3, just to see what
the figures are... maybe people who have reported freezing were using up
the memory ?

With the latest Raspbian release on the RPI3, at the CLI ( Command Line
Interface ), with no X running, a total 748 Megabytes of the 1 Gigabyte is
available for the operating system and applications. Of that, 647 Megabytes
is available for the applications.

With the window manager running, i.e. having issued a "startx" from the
CLI, 615 Megabytes is available for applications,

With the Qt-DAB fully running and all the windows of it open, 575 Megabytes
of memory are still free.

The good news is, then, that Qt-DAB is not freezing the system by using all
of the memory !

To do the test, if others would like to verify the above -

At just the CLI with no X running, issue the command  free -h
Then do a startx
When X is running, open a CLI window ( little black rectangle ) , and again
issue the command  free -h
Finally, start and run Qt-DAB ; when running, again issue the command  free
-h

I note that with the Qt-DAB compiled with cmake, running with Qt5 on the
latest Raspbian, that it is possible to "simulate" a "freeze". When Qt-DAB
first starts in my compilation, the "Dabstick" window is not displayed as
the dabstick is already allocated before the windows are active. If I click
on the "device" button on the main Qt-DAB window, and again select
"dabstick", the "frequency display" window freezes; I cannot change
settings on the main Qt-DAB window; ( I have only checked this when
changing from one DAB multiplex to another, so I don't know if audio
stopped too ). I was able to "unfreeze" the windows by clicking on the
green "start" button again. the white "reset" button did not cause an
"unfreeze". This paragraph may or may not be related to the findings of
other contributors ( above ) who have experienced loss of sound or
freezing. But I was able to resume the application, by clicking on the
green "start" button.

One last observation - with the latest Raspbian and Qt5, the processor
usage is now between 30 and 43 percent, so has slightly, but not
dramatically, increased.

Regards
Phil Dodd

JvanKatwijk, 2017-08-21:
> issuecomment-323677352
> +31 (0)15 3698980 <+31%2015%20369%208980> <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355> <+31%206%2028260355>

Thanks for the info. When selecting a device, basically the programs stops
until said that it should start again.
The comments on intermittent sound are hard to handle since in order to
handle I should be able to replay a section
with intermittent sound.
There are two possibilities: what often happens with dabsticks is that the
signal is weak and decoding some sections fail,
a problem would be if all data is correctly decoded and sound stops from
time to time.That as such could be an issue
of the program (i.e. samples get lost) or an issue of the environment, i.e.
portaudio/alsa looses or delays samples.

The isue is that I can not reconstruct such a situation on my RPI2

best
jan

Phil1951, 2017-08-22:
I'd like to mention another issue, just to hopefully eliminate it as a possible cause of dropouts. Settings of the Raspberry Pi are made through either the command "raspi-config", or from the menu item "Raspberry Pi Settings". One of the settings, under "advanced options", is to set the amount of memory allocated to the GPU. Some users found that the default value of 16 Megabytes was insufficient for what they were trying to do, so increased it to 256 Megabytes. I have followed that advice, and I set it to 256.

Perhaps if someone is using ALSA sound through HDMI, and only have 16 Megabytes allocated to the GPU, that could cause the sound to stop if the GPU runs out of memory ? Maybe in that case, try setting it to 256, just to eliminate that as a possible cause ?

JvanKatwijk, 2017-08-23:
Thanks for the tip.

Wrt the Raspbian. I just upgraded to the newest version, that version has
good support for Qt5 and the qwt6 compiled for Qt5.
Furtthermore, the version now supports GCC6.xx, a real improvement. I can
even compile the Ada version, since gtkada is supported,
but I do need an RPI3 to run it (overloads the RPI2 cores)

Phil1951, 2017-08-23:
Congratulations upon your excellent progress ! I am pleased to hear that
you have been able to resume work on the ADA version ! There are
conflicting stories about the RPI hardware versions. One magazine claimed
that a new, more powerful version was "in the pipeline". Other sources say
that the next version will not be released until 2019. It would seem
worthwhile, therefore, upgrading to an RPI 3B. If you are going to do so,
try to avoid the "Christmas rush" which can start as early as mid September.

Our radio and telecommunications body in the UK, Ofcom, used the RPI to
build a DAB transmitter for use by UK community groups, for "small scale
DAB transmissions". After initial experimentation, Ofcom, recommended the
Odroid XU4 as a more powerful alternative to the RPI - it has 8 cores, and
runs various Linux distributions including, I understand, Raspbian. Details
are here -

https://wiki.odroid.com/odroid-xu4/odroid-xu4

Regards

Phil Dodd

closed 2018-02-12 by JvanKatwijk