JvanKatwijk/qt-dab #66: Slideshows in appimage not working

andimik, 2017-09-15:
Seems the slideshows in the appimage do not work. I only get a very small window without the slide.

[image]

instead of the slide. The console output is

> going to show picture img01.jpg
> going to write file /tmp/qt-pictures/img01.jpg

The same release (d293dbf) built with Qt-Creator works well.

JvanKatwijk, 2017-09-16:
Interesting, the appImage is created in a pretty old Ubuntu environment
with Qt4.8. Next week I'll create the same Ubuntu env on
my old laptop, than I can see what the cause is
best
jan

JvanKatwijk, 2017-09-17:
This is interesting. What I did I installed a fresh Ubuntu 14.04 on my old
laptop. Then I installed everything needed to compile and run qt-dab and
ran qt-dab. Pictures were shown.

Then ...
I took the script for generating an appImage (used in the travis to
generate the appImage), created a separate subdirectory, adapted the script
and generated my own appImage. Pictures are NOT shown, when run on the
aforementioned Ubuntu environment.

Interesting, have not yet an answer

best
jan

ps
Your injuries healed in the meantime?
best
jan

2017-09-16 8:51 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Interesting, the appImage is created in a pretty old Ubuntu environment
> with Qt4.8. Next week I'll create the same Ubuntu env on
> my old laptop, than I can see what the cause is
> best
> jan
>
>
>
>> Seems the slideshows in the appimage do not work. I only get a very small
>> window without the slide.
>>
>> [image: grafik]
>> <https://user-images.githubusercontent.com/24510556/30495795-a1a70072-9a4d-11e7-9cbf-5a9c202da422.png>
>>
>> instead of the slide. The console output is
>>
>> going to show picture img01.jpg
>> going to write file /tmp/qt-pictures/img01.jpg
>>
>> The same release (d293dbf
>> <https://github.com/JvanKatwijk/qt-dab/commit/d293dbfca4a9cbf4de38e16b548bc9c62c0676a7>)
>> built with Qt-Creator works well.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

andimik, 2017-09-18:
Maybe it has to do with the picture geometry for the popup. 

Files are saved, BTW. 

Regarding the other topic I'll respond by private mail.

AyrA, 2017-11-28:
I can confirm, have the same issue using `qt-dab-1.0-alpha`.

Message:

    setGeometry: Unable to set geometry 5x13+640+280 on QWidgetWindow/'QLabelClassWindow'.
    Resulting geometry:  116x13+640+280 (frame: 8, 30, 8, 8, custom margin: 0, 0, 0, 0, minimum size: 0x0, maximum size: 16777215x16777215).

The jpg image is correctly written to disk however.

JvanKatwijk, 2017-11-28:
That is absolutely true. It has to do with the libraries in mingw for cross
compilation, but it is unclear which library. It is definitely a low level
library, since running under
Wine shows the pictures perfectly.

closed 2018-02-12 by JvanKatwijk