JvanKatwijk/qt-dab #67: RPI - Amendments of 21/09/2017 - epgdec.h and qdom.h

Phil1951, 2017-09-22:
Hi Jan and other followers !

Attempting to compile the snapshot as at 21/09/2017.

I have commented out rtl-sdr and airspy to leave dabstick, and attempting cmake using Qt5 on Raspbian Stretch.

At around 56% of the subsequent make compilation. in module src/backend/data/motdatabuilder.cpp an error is generated as it is unable to include epgdec.h ( from includes/backend/data/epg ).

I added the required lines into my CMakeList.txt to include the directory and header. The includes file once read, itself found an error, that it could not include qdom.h, which is in QtXml package. I tried including a findpackage for QtXml at the start of my CMakeList.txt, but this command was rejected by cmake.

I am sure that this is my lack of knowledge of cmake. I would be really grateful of any advice about the coding required to get QtXml discoverable by "make" at the above point.

Thank you for any comments

Phil

JvanKatwijk, 2017-09-26:
That is pretty strange, epgdec.h is included in mot-data.h, but only if
TRY_EPG is set, and I deliberately omitted the option
to set TRY_EPG in the CMakeLists.txt file, since that is experimental stuff

Can you send me the CMakeLists.txt file?

Phil1951, 2017-09-26:
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

Thank you for this comment. Before I send it, I will make another attempt,
without overwriting my last attempt - maybe I commented out too much - I do
not want to waste your time so I will double check and report back. Thank
you.

Phil1951, 2017-09-26:
Hello again.

I have recompiled from the September 22 qt-dab-master size 628454 using
cmake. This time, I did not comment anything out in the CMakeLists.txt, but
simply ran the compilation as

cmake CMakeLists.txt -DDABSTICK=ON -DSPECTRUM=ON

The compilation was totally successful this time and a test run of the
latest software is decoding the BBC multiplex on 12B in the UK, so all is
well.

I can only apologise to everyone for what has obviously been a false alarm
on my part, asking for help when it has been my own fault. In future, I
shall not comment anything out when updating this excellent piece of
software !

This exercise has not been a complete waste of time, though. I have
discovered to my delight that because the hidden file .qt-dab.ini is in the
"top level directory", it is available to different sub-directories
containing different versions of the software. This indeed a useful
time-saving feature ! Different versions run "out of the box" !

Thank you everyone ! unless anyone else has had issues compiling this
version, perhaps this thread can be closed ?

closed 2018-02-12 by JvanKatwijk