JvanKatwijk/qt-dab #68: No longer compiling on RPi3

NomisGninnac, 2017-10-03:
/usr/include/c++/4.9/atomic:642:15: error: expected ‘;’ at end of member declaration
       ~atomic() noexcept = default;
               ^
/usr/include/c++/4.9/atomic:642:17: error: ‘noexcept’ does not name a type
       ~atomic() noexcept = default;
                 ^
/usr/include/c++/4.9/atomic:647:7: error: ‘constexpr’ does not name a type
       constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
       ^
/usr/include/c++/4.9/atomic:647:7: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/include/c++/4.9/atomic:660:14: error: expected ‘;’ at end of member declaration
       atomic() noexcept = default;
              ^
/usr/include/c++/4.9/atomic:660:16: error: ‘noexcept’ does not name a type
       atomic() noexcept = default;
                ^
/usr/include/c++/4.9/atomic:661:15: error: expected ‘;’ at end of member declaration
       ~atomic() noexcept = default;
               ^
/usr/include/c++/4.9/atomic:661:17: error: ‘noexcept’ does not name a type
       ~atomic() noexcept = default;
                 ^
/usr/include/c++/4.9/atomic:666:7: error: ‘constexpr’ does not name a type
       constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
       ^
/usr/include/c++/4.9/atomic:666:7: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/include/c++/4.9/atomic:679:14: error: expected ‘;’ at end of member declaration
       atomic() noexcept = default;
              ^
/usr/include/c++/4.9/atomic:679:16: error: ‘noexcept’ does not name a type
       atomic() noexcept = default;
                ^
/usr/include/c++/4.9/atomic:680:15: error: expected ‘;’ at end of member declaration
       ~atomic() noexcept = default;
               ^
/usr/include/c++/4.9/atomic:680:17: error: ‘noexcept’ does not name a type
       ~atomic() noexcept = default;
                 ^
/usr/include/c++/4.9/atomic:685:7: error: ‘constexpr’ does not name a type
       constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
       ^
/usr/include/c++/4.9/atomic:685:7: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/include/c++/4.9/atomic:698:14: error: expected ‘;’ at end of member declaration
       atomic() noexcept = default;
              ^
/usr/include/c++/4.9/atomic:698:16: error: ‘noexcept’ does not name a type
       atomic() noexcept = default;
                ^
/usr/include/c++/4.9/atomic:699:15: error: expected ‘;’ at end of member declaration
       ~atomic() noexcept = default;
               ^
/usr/include/c++/4.9/atomic:699:17: error: ‘noexcept’ does not name a type
       ~atomic() noexcept = default;
                 ^
/usr/include/c++/4.9/atomic:704:7: error: ‘constexpr’ does not name a type
       constexpr atomic(__integral_type __i) noexcept : __base_type(__i) { }
       ^
/usr/include/c++/4.9/atomic:704:7: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/include/c++/4.9/atomic:717:14: error: expected ‘;’ at end of member declaration
       atomic() noexcept = default;
              ^
/usr/include/c++/4.9/atomic:717:16: error: ‘noexcept’ does not name a type
       atomic() noexcept = default;
                ^
/usr/include/c++/4.9/atomic:718:15: error: expected ‘;’ at end of member declaration
       ~atomic() noexcept = default;
               ^
/usr/include/c++/4.9/atomic:718:17: error: ‘noexcept’ does not name a type
       ~atomic() noexcept = default;
                 ^
[... 190 lines cut ...]
/usr/include/c++/4.9/atomic:1103:43: error: expected initializer before ‘noexcept’
          ptrdiff_t __d, memory_order __m) noexcept
                                           ^
/usr/include/c++/4.9/atomic:1109:28: error: expected initializer before ‘noexcept’
          memory_order __m) noexcept
                            ^
/usr/include/c++/4.9/atomic:1114:66: error: expected initializer before ‘noexcept’
     atomic_fetch_sub(volatile atomic<_ITp*>* __a, ptrdiff_t __d) noexcept
                                                                  ^
/usr/include/c++/4.9/atomic:1119:57: error: expected initializer before ‘noexcept’
     atomic_fetch_sub(atomic<_ITp*>* __a, ptrdiff_t __d) noexcept
                                                         ^
In file included from includes/backend/fic-handler.h:34:0,
                 from dab-processor.h:37,
                 from radio.h:40,
                 from main.cpp:33:
includes/backend/fib-processor.h:94:4: warning: anonymous type with no linkage used to declare variable ‘<anonymous struct> ensemble_Descriptor’ with linkage
  } ensemble_Descriptor;
    ^
main.cpp: In function ‘int main(int, char**)’:
main.cpp:63:17: warning: unused variable ‘tracing’ [-Wunused-variable]
 bool            tracing         = false;
                 ^
Makefile:584: recipe for target 'main.o' failed
make: *** [main.o] Error 1

NomisGninnac, 2017-10-04:
Just tried again with newest but exactly the same error

NomisGninnac, 2017-10-04:
pi@raspberrypi:~/qt-dab $ qmake qt-dab.pro-linux
WARNING: Failure to find: src/ofdm/ofdm-processor.cpp
WARNING: Failure to find: src/backend/dab-processor.cpp
WARNING: Failure to find: includes/ofdm/ofdm-processor.h
WARNING: Failure to find: includes/backend/dab-processor.h
pi@raspberrypi:~/qt-dab $

NomisGninnac, 2017-10-04:
Have tried all possible options.  qt-dab.pro and qt-dab.pro-linux. 
This project is currently dead on RPi3.
Over 20 compilations attempts or so.
Please look into this as its important

JvanKatwijk, 2017-10-04:
It seems you are running with a 4.9 C++ compiler. I am running stretch with
a 6.X compiler and have no problems on the RPI2
It might be the case that 4.9 needs includes where 6.X takes them as given.
Can you be more specific on the file where the error occurs

JvanKatwijk, 2017-10-04:
You might try adding

CONFIG += c++11

to the ".pro" file

NomisGninnac, 2017-10-04:
Have now done the following:

Formatted SD Card and updated to latest stretch
Compiled successfully.
QT-DAB can now no longer detect Dabstick

Yet, lsusb says this;
Bus 001 Device 011: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

JvanKatwijk, 2017-10-04:
is the udev setting ok?

NomisGninnac, 2017-10-04:
I do not know

JvanKatwijk, 2017-10-04:
well you refreshed your system, then - when creating a library for the
dabstick - you should also install a file in the directory
/etc/udev/rules.d to ensure that non-root users can access the usb device

NomisGninnac, 2017-10-04:
I have no idea how to do that.  I've never had a problem with the last 18 or so releases.

NomisGninnac, 2017-10-04:
running

rtl_test  -t

gives:

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.
Reattached kernel driver

andimik, 2017-10-04:
Let me tell you that I had no problems building qt-dab on my Rpi 3 around 1 month ago. But I have not tested rtl_sdr as I am using an Airspy normally. 

With the latest Raspbian also the spectrum can be used, too.

NomisGninnac, 2017-10-04:
I set all RTL-SDR selections to reinstall using synaptic

This has allowed me to launch QT-DAB.  Channels are showing, but the moment I click on anything, including "quit" everything seizes.

NomisGninnac, 2017-10-04:
I didn't think to backup my old sd-card before formatting for the new Debian Stretch.  I was too excited to get the alpha qt-dab...
So yes, a month ago was about right. I'm kicking myself that I didn't backup. 

I will wait until a few more updates come through and see if others have similar problems
I love this program, I will stick by it.

andimik, 2017-10-05:
You always can download older versions. This is the benefit of github.com 

Just go to the commits page and you see a <> button on the right.

JvanKatwijk, 2017-10-05:
What is the error message you get when selecting a dabstick

NomisGninnac, 2017-10-05:
[image]
Have created rules file in udev, rebooted etc, but still getting blocked.

JvanKatwijk, 2017-10-06:
What error message do you get on a terminal window. What OS you are running

NomisGninnac, 2017-10-06:
[image]
Have reflashed SD Card since last attempt.  Have installed RTL-SDR again.  Still having sporadic crashing.  Audio has stopped working in the last few minutes without changing anything.  Muting then unmuting audio on the Desktop brings it back however the audio is choppy.

No discernible errors written in console.   Am trying to work out how to get an older version, about a month ago was OK.

Console output:
added items to combobox 
Suggested size for outputbuffer = 2560
stream opened
stream started
4 dabstick
OK, functions seem to be loaded
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 49.6 48.0 44.5 43.9 43.4 42.1 40.2 38.6 37.2 36.4 33.8 32.8 29.7 28.0 25.4 22.9 20.7 19.7 16.6 15.7 14.4 12.5 8.7 7.7 3.7 2.7 1.4 0.9 0.0 
Starting 1

Selected service talkRADIO       
we now have DAB
Suggested size for outputbuffer = 2560
stream opened
stream started
Suggested size for outputbuffer = 2560
stream opened
stream started
dabProcessor is stopping
Selected service BBC Radio 4     
we now have DAB
dabProcessor is stopping
Selected service talkRADIO       
we now have DAB

JvanKatwijk, 2017-10-06:
Did you uncomment THREADED_DECODING?  In case you did not, mos computations
are done in a single thread which - at least
on the RPI 2 - overloads one of the cores

NomisGninnac, 2017-10-06:
Unable to compile with it uncommented.

Console:
In file included from src/backend/audio/dab-audio.cpp:24:0:
includes/backend/audio/dab-audio.h:61:2: error: ‘atomic’ does not name a type
  atomic<bool> running;
  ^~~~~~
src/backend/audio/dab-audio.cpp: In constructor ‘dabAudio::dabAudio(RadioInterface*, uint8_t, int16_t, int16_t, bool, int16_t, RingBuffer<short int>*, QString)’:
src/backend/audio/dab-audio.cpp:96:2: error: ‘running’ was not declared in this scope
  running. store (true);
  ^~~~~~~
src/backend/audio/dab-audio.cpp:47:12: warning: unused variable ‘j’ [-Wunused-variable]
 int32_t i, j;
            ^
src/backend/audio/dab-audio.cpp: In destructor ‘virtual dabAudio::~dabAudio()’:
src/backend/audio/dab-audio.cpp:104:2: error: ‘running’ was not declared in this scope
  running. store (false);
  ^~~~~~~
src/backend/audio/dab-audio.cpp: In member function ‘virtual int32_t dabAudio::process(int16_t*, int16_t)’:
src/backend/audio/dab-audio.cpp:125:10: error: ‘running’ was not declared in this scope
     if (!running)
          ^~~~~~~
src/backend/audio/dab-audio.cpp:121:48: warning: unused parameter ‘cnt’ [-Wunused-parameter]
 int32_t dabAudio::process (int16_t *v, int16_t cnt) {
                                                ^~~
src/backend/audio/dab-audio.cpp: In member function ‘virtual void dabAudio::run()’:
src/backend/audio/dab-audio.cpp:178:9: error: ‘running’ was not declared in this scope
  while (running. load ()) {
         ^~~~~~~
src/backend/audio/dab-audio.cpp: In member function ‘virtual void dabAudio::stopRunning()’:
src/backend/audio/dab-audio.cpp:190:2: error: ‘running’ was not declared in this scope
  running = false;
  ^~~~~~~
Makefile:888: recipe for target 'dab-audio.o' failed
make: *** [dab-audio.o] Error 1

JvanKatwijk, 2017-10-06:
> atomic running;

what system do you runthen

JvanKatwijk, 2017-10-06:
apparently you are running a system with a pretty old gcc that does not
support c++11 automatically

Better go back to an older version of qt-dab since I am increasing the use
of c++11 in my software (after all c++17 is almost here)

2017-10-06 22:27 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> what system do you runthen
>
>
>
>> Unable to compile with it uncommented.
>>
>> Console:
>> In file included from src/backend/audio/dab-audio.cpp:24:0:
>> includes/backend/audio/dab-audio.h:61:2: error: ‘atomic’ does not name a
>> type
>> atomic running;
>> ^~~~~~
>> src/backend/audio/dab-audio.cpp: In constructor
>> ‘dabAudio::dabAudio(RadioInterface*, uint8_t, int16_t, int16_t, bool,
>> int16_t, RingBuffer
>>
>>
>>
>>
>>
>>
>>
>>
>>
>>
>> *, QString)’: src/backend/audio/dab-audio.cpp:96:2: error: ‘running’ was
>> not declared in this scope running. store (true); ^~~~~~~
>> src/backend/audio/dab-audio.cpp:47:12: warning: unused variable ‘j’
>> [-Wunused-variable] int32_t i, j; ^ src/backend/audio/dab-audio.cpp: In
>> destructor ‘virtual dabAudio::dabAudio()’:
>> src/backend/audio/dab-audio.cpp:104:2: error: ‘running’ was not declared in
>> this scope running. store (false); ^ src/backend/audio/dab-audio.cpp: In
>> member function ‘virtual int32_t dabAudio::process(int16_t*, int16_t)’:
>> src/backend/audio/dab-audio.cpp:125:10: error: ‘running’ was not
>> declared in this scope
>> if (!running)
>> ^~~~~~~
>> src/backend/audio/dab-audio.cpp:121:48: warning: unused parameter ‘cnt’
>> [-Wunused-parameter]
>> int32_t dabAudio::process (int16_t *v, int16_t cnt) {
>> ^~~
>> src/backend/audio/dab-audio.cpp: In member function ‘virtual void
>> dabAudio::run()’:
>> src/backend/audio/dab-audio.cpp:178:9: error: ‘running’ was not declared
>> in this scope
>> while (running. load ()) {
>> ^~~~~~~
>> src/backend/audio/dab-audio.cpp: In member function ‘virtual void
>> dabAudio::stopRunning()’:
>> src/backend/audio/dab-audio.cpp:190:2: error: ‘running’ was not declared
>> in this scope
>> running = false;
>> ^~~~~~~
>> Makefile:888: recipe for target 'dab-audio.o' failed
>> make: *** [dab-audio.o] Error 1
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

NomisGninnac, 2017-10-06:
Raspian Stretch 7th September Release.

I have retracted to Commits on Sep 21, 2017, works pretty well

JvanKatwijk, 2017-10-07:
OK

Interesting that you have problems withthe recent version. I run it on the
RPI2 with stetch without any problem whatsoever
But we can close this issue I presume

NomisGninnac, 2017-10-07:
Yes, you may close the issue.  Although I'd like to work out why the GCC compilers aren't up to scratch or allowing compilation of latest code.  This isn't down to you though, as it's a Pi3 library issue.

closed 2017-10-07 by NomisGninnac