JvanKatwijk/qt-dab #69: Use QApplication::setWindowIcon() to set app icon

probonopd, 2017-11-18:
Please use `QApplication::setWindowIcon()` so that the application, when launched from the AppImage, will show an icon in the dock.

[image]

JvanKatwijk, 2017-11-19:
Actually I did this. It seems I get a nice icon when compiling with
qmake/make, but nothing happens when compiling
using cmake (as is done in the travis route for the appImage)

Any suggestion will be appreciated

probonopd, 2017-11-19:
Maybe cmake is not installing the icon in the correct location?

You could try to run both versions with `strace`, and check for differences.

JvanKatwijk, 2017-11-20:
afaik solved. Adding setting the resources compiler in the CMakeLists.txt file did the trick

closed 2017-11-20 by JvanKatwijk