JvanKatwijk/qt-dab #75: MOD Slideshow in Windows

veso266, 2017-12-28:
Hi I am using latest (compiled) version of qt-dab from here: https://www.sdr-j.tk/windows-bin.zip and I was always wanted to see MOT Slideshows but was never able to find an option to display them
as you can see here: https://www.youtube.com/watch?v=VQqKGjIrZy0&feature=youtu.be the MOT Slideshow is decoded but I am never able to find an image (and I know that Croatia DAB Stations have MOT Slideshow (I saw them here: https://www.youtube.com/watch?v=NrBTWDXK4ik) I also tried with Slovenian DAB+ station that has MOT Slideshow too but I was never able to see the image

Thanks for Anwsering and Best Regards
PS: Do you know if is posible to somehow improve Audio decoding a bit on a weak stations (because Esemble info shows with very weak signal (and station names show too) but you need a realy strong signal to get Audio out of it and for some reason only qt-dab is able to decode audio from these stations here: https://www.youtube.com/watch?v=VQqKGjIrZy0 (I was never able to get Audio to decode when using sdr-j-dab-0.9985 (not sure what has changed)
PPS: will there be a windows version of only Command Line version of DAB Decoder?

andimik, 2017-12-28:
Good evening, I know both ensembles (from SLO and HRV) very well and am in close contact with OiV and RTV Slovenija.

Currently there is a Windows problem with the slides. Please close the very small slideshow window (you will have to resize it for closing). Then it may happen that the slide is displayed correctly.

If you can, please try Qt-DAB under Linux.

Please check VAL202, which often has very small slides (1 kB approx.). Radio Slovenia International has the same 2 slides alternating all the time. Slides for all private stations from DAB 10D really take very long (file size is large and the bitrate for data is reduced). So please try RTV public channels first.

You are right, for Croatian OiV test transmissions, slides are not displayed. But this has to do because of the next paragraph:
 
Nevertheless, slideshows in Qt-DAB only work when the data are received continuously in the MOT stream without other data in between (= the green MOT indicator is always green without interruption). See https://github.com/JvanKatwijk/qt-dab/issues/21

andimik, 2017-12-28:
The ini file should be located in c:\users\[user] under Windows.

veso266, 2017-12-28:
Thanks 
Found the ini file in %userprofile%\qt-dab.ini

I was able to get only the slide window displayed without picture (only on Slo1, and ARS)

I wish someone would make DAB Player compatible with AirSpy R2

andimik, 2017-12-29:
Seems you are not up to date. Andreas Gsinn's player works with Airspy. First you have to start the interface program and then his player using rtltcp parameter.

veso266, 2017-12-29:
you are right Thank

andimik, 2017-12-29:
qt-sdr is no more existing ;-)

It depends on what you want to do with the DAB stream.

# Andi Gsinn's player
programmer with best knowledge, multi language
supports RTLSDR (but NOT Zadig), HackRF and Airspy and rtltcp, but no prerecorded files so far (but to be announced)
Saves screenshots, 
CU usage, 
many graphs, 
audio output, 
allows export of config for DABLIST :+1: 
Best for DAB, but only under Windows or Wine

# Qt-DAB
has more inputs (+ SDRplay), 
reads from raw and (own) sdr files, 
exports sdr (not raw), 
reads TII in FIC (coordinates), but not from Null Symbols, 
TPEG and EPG (but still under development, does not work with me),
is a bit buggy (sorry Jan)

# dablin 
only supports ETI as input (from file or via pipe), 
very stable, 
programmer works for DAB software in real live

# welle.io
supports Airspy, RTLSDR, raw, sdr (16be and 16le). 
Only user friendly implementation (= not deep), 
very stable
for Windows, Linux, Mac and Android
Multilingual 

# Fraunhofer
only for Windows, 
very old

# DAB Player from Realtek
only for Windows, 
very, very old (= the grandfather of all programs)
has raw output, 
audio output
Remark: I made raw files from old DAB Slovenija even in 2012 with it (and re-discovered them by chance some days ago)

# Qirx
Windows, 
brandnew, 
but still very buggy, 
the only program for TII in Null Symbols
has raw input

# SDRDAB 
from Poland, v3beta, still under development

JvanKatwijk, 2018-01-02:
> v=VQqKGjIrZy0&feature=youtu.be the MOT Slideshow is decoded but I am

Thanks for the message

First of all, you better use the most recent version of qt-dab, which is to
be found on https://github.com/JvanKatwijk/qt-dab
It contains some improvements. I have to look into the locations where the
slides are stored, windows always gives problems,
I am not a fan of it. There were some other complaints of not having
pictures stored. Afaik they are stored in a %tmp directory

It is certainly possible to have a cmdline version cross compiled, but it
is complicated since - obviously - all choices have to be
compiled in. So, rather than having one executable with selectors on it,
one ends up with app 16 versions.

Looking at your movies, the signal is quite weak (forming DAB+ frames fails
quite often)

best
jan

andimik, 2018-01-29:
We could close that I suppose...

JvanKatwijk, 2018-01-30:
OK with me

closed 2018-02-12 by JvanKatwijk