JvanKatwijk/qt-dab #77: zip file from github.com not supported

andimik, 2018-01-01:
If a user downloads the official zip here from https://github.com/JvanKatwijk/qt-dab/archive/master.zip , it does not compile as there is no hidden `.git` directory existing in the zip file.

# Using cmake

```
andreas@andiE535:~/qt-dab-master/build$ cmake .. -DAIRSPY=ON
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
fatal: Not a git repository (or any of the parent directories): .git
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found zlib: 
-- Found LibSndFile: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Found LibSampleRate: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Checking for module 'portaudio-2.0'
--   Found portaudio-2.0, version 19
-- Performing Test PORTAUDIO2_FOUND
-- Performing Test PORTAUDIO2_FOUND - Success
-- Found PORTAUDIO: /usr/include  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andreas/qt-dab-master/build
```

and

```
/home/andreas/qt-dab-master/main.cpp: In function ‘int main(int, char**)’:
/home/andreas/qt-dab-master/main.cpp:73:82: error: ‘GITHASH’ was not declared in this scope
 cation::setApplicationVersion (QString (CURRENT_VERSION) + " Git: " + GITHASH);
                                                                       ^
/home/andreas/qt-dab-master/main.cpp:65:17: warning: unused variable ‘tii_delay’ [-Wunused-variable]
 int16_t         tii_delay       = 20;
                 ^
CMakeFiles/qt-dab-1.0-alpha.dir/build.make:178: die Regel für Ziel „CMakeFiles/qt-dab-1.0-alpha.dir/main.cpp.o“ scheiterte
make[2]: *** [CMakeFiles/qt-dab-1.0-alpha.dir/main.cpp.o] Fehler 1
CMakeFiles/Makefile2:67: die Regel für Ziel „CMakeFiles/qt-dab-1.0-alpha.dir/all“ scheiterte
make[1]: *** [CMakeFiles/qt-dab-1.0-alpha.dir/all] Fehler 2
Makefile:127: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```

# Using qmake

```
andreas@andiE535:~/qt-dab-master/build$ qmake ..
fatal: Not a git repository (or any of the parent directories): .git
```

It would make sense to check if the directory exists and if not, then use default or no values.

JvanKatwijk, 2018-01-01:
> JvanKatwijk/qt-dab/archive/master.zip , it does not compile as there is

It took some digging into qmake/cmake syntax, but is seems corrected.
Thanks for the observation

andimik, 2018-01-03:
Yes, confirmed. Now it is working.

Tested with https://github.com/JvanKatwijk/qt-dab/commit/671cdb237f16b8d813930a897085af151200fe43

closed 2018-01-03 by andimik