JvanKatwijk/qt-dab #78: Versionizing

andimik, 2018-01-01:
The version is now minus (sic!) 1.0 alpha

see `#define	CURRENT_VERSION	"-1.0 alpha"`in `dab-constants.h`

Furthermore, the sorting in the "About window" is very strange: 
Weekday MMM DD HH:MM:SS YYYY build

Why not Weekday MMM DD YYYY HH:MM:SS build?

```
//	display the version
	QString v = "Qt-DAB " + QString (CURRENT_VERSION);
	QString versionText = "qt-dab version: " + QString(CURRENT_VERSION);
        versionText += "Build on: " + QString(__TIMESTAMP__) + QString (" ") + QString (GITHASH);
	versionName	-> setText (v);
	versionName	-> setToolTip (versionText);
```

I could not find out, where "TIMESTAMP" is defined. There is only one string found in radio.cpp, see code above.

[image]

JvanKatwijk, 2018-01-01:
Hi Andreas

First of all, let us make it into a good year. Thanks for the comments on
detail you defintely have an eye for it, I do not.
The TIMESTAMP is a macro filled in by the compiler, I do not know (yet?)
how to change that setting

The version is now made positive again, the '-' is attached to the Qt-DAB
string preceding it in the GUI.

best
jan

andimik, 2018-01-04:
Would this help?

https://cmake.org/cmake/help/v3.0/command/string.html

closed 2018-02-12 by JvanKatwijk