JvanKatwijk/qt-dab #79: RAW files need write at least r-x (optical medium not supported)

andimik, 2018-01-03:
Hi Jan,

I found a very interesting bug by checking my raw file archive on DVD. **RAW files on an optical medium (CD, DVD) are not supported, as they normally don't have execution attribute.** 

BTW: Welle.io opens the same rawfile on DVD without problems. So it has to do with your code.

SDR files are not affected, they even work with r--

qt-dab, eti-cmdline-rawfiles and dab-raw-2 have the same problem.

[image]

[image]

```
andreas@andiE535:~$ mount | grep DAB
/dev/sr0 on /media/andreas/DAB_SVK12C_Feb17 type iso9660 (ro,nosuid,nodev,relatime,uid=1000,gid=1000,iocharset=utf8,mode=0400,dmode=0500,uhelper=udisks2)
```

## raw file stored on a DVD

```
andreas@andiE535:~$ eti-cmdline-rawfiles -F /media/andreas/DAB_SVK12C_Feb17/12c_pressburg2.raw > 12c_pressburg2.eti
could not open /media/andreas/DAB_SVK12C_Feb17/12c_pressburg2.raw
Installing device failed, fatal21

andreas@andiE535:~$ dab-raw-2 -F /media/andreas/DAB_SVK12C_Feb17/12c_pressburg2.raw
dab_cmdline V 1.0alfa,
 	                  Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = F
file /media/andreas/DAB_SVK12C_Feb17/12c_pressburg2.raw cannot open
allocating device failed (31), fatal

```

## same file copied to home dir

```
andreas@andiE535:~$ eti-cmdline-rawfiles -F ~/12c_pressburg2.raw > 12c_pressburg2.eti
Period = 8000
ensemble Towercom detected
program ( 1)	 Radio Slovensko	 5000 is in the list
program ( 2)	 Radio Patria	 5001 is in the list
program ( 3)	 Radio Junior	 5002 is in the list
program ( 4)	 Radio_FM	 5003 is in the list
program ( 5)	 Radio Devin	 521B is in the list
program ( 6)	 Radio Litera	 5005 is in the list
there might be a DAB signal here

andreas@andiE535:~$ dab-raw-2 -F 12c_pressburg2.rawdab_cmdline V 1.0alfa,
 	                  Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = F
[...]
Suggested size for outputbuffer = 1280
ofdm word gestart
Period = 8000
program Radio Slovensko is part of the ensemble
ensemble Towercom is (5001) recognized

```

## SDR file on DVD works

```
andreas@andiE535:~$ eti-cmdline-wavfiles -F /media/andreas/DAB_SVK12C_Feb17/12c_pressburg2.sdr > 12c_pressburg2.eti
Period = 16000
there might be a DAB signal here
ensemble Towercom detected wait
program ( 1)	 Radio Slovensko	 5000 is in the list
program ( 2)	 Radio Patria	 5001 is in the list
program ( 3)	 Radio Junior	 5002 is in the list
program ( 4)	 Radio_FM	 5003 is in the list
program ( 5)	 Radio Devin	 521B is in the list
program ( 6)	 Radio Litera	 5005 is in the list
```

## content of the DVD

```
andreas@andiE535:/media/andreas/DAB_SVK12C_Feb17$ ls -l
[...]
-r--r--r-- 1 andreas andreas  472846358 Jul 31 12:35 12c_pressburg2.raw
-r-xr-xr-x 2 andreas andreas  945692716 Feb 25  2017 12c_pressburg2.sdr
```

The only thing is that they have no execute attribute. When I set it to r-- then also the home directory file cannot played.

JvanKatwijk, 2018-01-03:
It turned out that "r+b" in the fopen apparently assumes that you want to
write on the device, while "rb" does not.
Changed
j

andimik, 2018-01-04:
Yes, thanks. Now it works :+1: 

Fixed in https://github.com/JvanKatwijk/qt-dab/commit/c3cf6afb2742828de4fe31255887e7def8681c4a

closed 2018-01-04 by andimik