JvanKatwijk/qt-dab #81: EPG issues - Variable fprintf

andimik, 2018-01-04:
I don't understand what you wanted to do with this code in `radio.cpp`, lines 1259ff

```
	   case PACKET_SERVICE:
	      {  packetdata d;
	         my_dabProcessor -> dataforDataService (s, &d);
	         if ((d.  DSCTy == 0) || (d. bitRate == 0)) {
	            fprintf (stderr, "d. DSCTy = %d, d. bitRate = %d\n",
	                               d. DSCTy, d. bitRate);
	            QMessageBox::warning (this, tr ("sdr"),
 	                               tr ("still insufficient data for this service\n"));

	            return;
	         }
```

Every time I click on an EPG data service, the bitRate is displayed as 0, not as actual kBit/s. So 0 is the same value you've specified one line above ...

For example 5C Bundesmux from Germany

> Selected service EPG Deutschland  (data)
> d. DSCTy = 0, d. bitRate = 0

BTW: try-epg is configured in the `qt-dab.pro`-file. But I have newer seen that an `*.ehb` file has been written.

In former versions (for example 1st of April 2017, see https://github.com/JvanKatwijk/qt-dab/tree/2523434fc9f579cd679a60898bbdac6ee90ff790 ) the small logos from the EPGs were shown and saved to qt-pictures. This does not work any more, unfortunately. 

Here from Vienna:

> AC51_ENERGY_112.png
> AC51_ENERGY_ENERGY_ENERGY.png
> AC51_ENERGY_neu.png
> AC51.png
> AC52_klassik_radio klassik_radio klassik_32x32.png
> AC52_radio_128.png
> AC52_radio_neu.png
> AC52_radio.png
> AC66_LoungeFM_112.png
> AC66_LoungeFM_128x128.png
> AC66_LoungeFM_LoungeFM_LoungeFM_32x32.png
> AC66_LoungeFM_neu.png
> AC67_Maria_Radio Maria_Radio Maria_32x32.png
> AD01_Melodie_Radio Melodie_Radio Melodie_32x32.png
> AD01_Radio_Melodie_128.png
> AD03_12.png
> AD03_ERF_ERF Plus_ERF Plus_32x32.png
> AD03.png
> AD05_112.png
> AD05_128.png
> AD05_HEROLD_HEROLD relax_HEROLD relax_32x32.png
> AD06_MEGA_MEGA Radio_MEGA Radio_32x32.png
> AD08_Radio_Technikum_112.png
> AD08_Radio_Technikum_128.png
> AD08_Radio_Technikum_neu.png
> AD09_Welle_1_112.png
> AD09_Welle_1_128.png
> AD09_Welle_1_neu.png
> AD09_WELLE 1_WELLE 1 DIGITAL_WELLE 1 DIGITAL_32x32.png
> AD0B_ARBÖ_32x32.png
> AD0B_ARBÖ_ARBÖ-Radio_ARBÖ-Radio_112.png
> adA0_128.png
> ada0_Arabella_112x32.png
> ada0_Arabella_Arabella Rock_Arabella Rock_32x32.png

JvanKatwijk, 2018-01-05:
> JvanKatwijk/qt-dab/tree/2523434fc9f579cd679a60898bbdac6ee90ff790 ) the

Andreas

The code is just to prevent an insufficiently specified service to blow up
code when ececuting

It would be very helpful if you can create a dumpfile that does show things
with the april 1 version and does not show data with the current version,
I do not have decent dumps with a sufficient amount of EPG data. With the
TPG data (that one we do have here) there is no difference

best
jan

andimik, 2018-01-08:
Thanks, the new version now shows the pics again. Testet with rawfiles from Vienna and also SWR from Germany. The EHB file is only written on the SWR ensemble. Not in Slovenia, not in Germany 5C Bundesmux. 

Rawfiles are here http://radioforum.foren.mysnip.de/read.php?35326,1452039,1452160#msg-1452160

andimik, 2018-02-09:
# Good news: 

## 1.  You have fixed the line 

> Selected service EPG Deutschland (data)
> d. DSCTy = 0, d. bitRate = 0

for example 5C Bundesmux now shows

> startAddr 444, Length 12, bitrate 16, DSCTy 60, appType 0

## 2. and 5C gets now EHB files and EPG pictures :-)

> startAddr 444, Length 12, bitrate 16, DSCTy 60, appType 0
> epgdata being handled
> filename = /tmp/qt-pictures/w20180103dd230c0.EHB
> going to write file /tmp/qt-pictures/Dlf_128x128.png
> going to write file /tmp/qt-pictures/Dlf_32x32.png
> going to write file /tmp/qt-pictures/Dlf_DokDeb_112x32.png
> going to write file /tmp/qt-pictures/Dlf_DokDeb_128x128.png
> going to write file /tmp/qt-pictures/Dlf_DokDeb_32x32.png
> going to write file /tmp/qt-pictures/Dlf_Kultur_112x32.png
> going to write file /tmp/qt-pictures/Dlf_Kultur_128x128.png
> going to write file /tmp/qt-pictures/Dlf_Kultur_32x32.png
> going to write file /tmp/qt-pictures/Dlf_Nova_112x32.png
> going to write file /tmp/qt-pictures/Dlf_Nova_128x128.png
> epgdata being handled
> End of file, restarting

## 3. And the EPG Pictures now are more (I used the same raw file recorded in Vienna last year)

### Version a27623d

> Selected service DAB Wien EPG-ORS
> data: protlevel = 3
> going to write file /tmp/qt-pictures/img/AD08_Radio_Technikum_112.png
> going to write file /tmp/qt-pictures/img/AD08_Radio_Technikum_128.png
> going to write file /tmp/qt-pictures/img/AD08_Radio_Technikum_neu.png
> going to write file /tmp/qt-pictures/img/AD08_Technik_Radio Technikum_Radio Te
> going to write file /tmp/qt-pictures/img/AD09_WELLE 1_WELLE 1 DIGITAL_WELLE 1 DIGITAL_32x32.png
> going to write file /tmp/qt-pictures/img/AD09_Welle_1_112.png
> going to write file /tmp/qt-pictures/img/AD09_Welle_1_128.png
> going to write file /tmp/qt-pictures/img/AD09_Welle_1_neu.png
> going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_32x32.png
> going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_ARBÖ-Radio_ARBÖ-Radio_112.png
> going to write file /tmp/qt-pictures/img/AC52_radio_neu.png
> End of file, restarting

### Version from today e57d8ba

> going to write file /tmp/qt-pictures/img/AD05_112.png
> going to write file /tmp/qt-pictures/img/AD05_128.png
> going to write file /tmp/qt-pictures/img/AD05_HEROLD_HEROLD relax_HEROLD relax_32x32.png
> going to write file /tmp/qt-pictures/img/AD06_MEGA_MEGA Radio_MEGA Radio_32x32.png
> going to write file /tmp/qt-pictures/img/AD06_MEGA_Radio_112.png
> going to write file /tmp/qt-pictures/img/AD08_Radio_Technikum_112.png
> going to write file /tmp/qt-pictures/img/AD08_Radio_Technikum_128.png
> going to write file /tmp/qt-pictures/img/AD08_Technik_Radio Technikum_Radio Te
> going to write file /tmp/qt-pictures/img/AD09_WELLE 1_WELLE 1 DIGITAL_WELLE 1 DIGITAL_32x32.png
> going to write file /tmp/qt-pictures/img/AD09_Welle_1_128.png
> going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_32x32.png
> going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_ARBÖ-Radio_ARBÖ-Radio_112.png
> going to write file /tmp/qt-pictures/img/AC51_ENERGY_112.png
> going to write file /tmp/qt-pictures/img/AC51_ENERGY_neu.png
> going to write file /tmp/qt-pictures/img/AC52_radio.png
> going to write file /tmp/qt-pictures/img/AC66_LoungeFM_128x128.png
> going to write file /tmp/qt-pictures/img/AC66_LoungeFM_LoungeFM_LoungeFM_32x32.png
> going to write file /tmp/qt-pictures/img/AC66_LoungeFM_neu.png
> going to write file /tmp/qt-pictures/img/AC67_Maria_Radio Maria_Radio Maria_32x32.png
> going to write file /tmp/qt-pictures/img/AC67_Radio_Maria_112.png
> going to write file /tmp/qt-pictures/img/AC67_Radio_Maria_128.png
> going to write file /tmp/qt-pictures/img/AD01_Melodie_Radio Melodie_Radio Melodie_32x32.png

Good achievements!

Still waiting for EHB for Vienna and Slovenia and RAI.

closed 2018-02-12 by JvanKatwijk