JvanKatwijk/qt-dab #82: TII Main ID errors for carriers above -700

tghe-retford, 2018-01-06:
With the recent work on TII implementation within QT-DAB, I have been keeping an keen interest on the development of TII decoding for the main and sub ID's for Linux. One thing I have noticed with recent changes is that for carriers above -700, the main ID is incorrect.

This is an example from the Lincolnshire multiplex. On my set-up, I should get a main ID of 47 (Humberside) and sub ID of 7 (High Hunsley) which is correct at carrier -754. But for carrier -658, the main ID fluctuates - in the examples below the main ID's of 8, 5 and 11 translate to London and the Home Counties (8 and 11) and North and West Wales (5):

```
the pattern is 211421142114211 at carrier -658
guess is: mainId 8 (0x8), subId 7 (0x7)
the pattern is 211421142114211 at carrier -658
guess is: mainId 8 (0x8), subId 7 (0x7)
the pattern is 231223122312231 at carrier -754
guess is: mainId 47 (0x2F), subId 7 (0x7)
the pattern is 311331133113311 at carrier -658
guess is: mainId 5 (0x5), subId 7 (0x7)
the pattern is 311331133113311 at carrier -658
guess is: mainId 5 (0x5), subId 7 (0x7)
the pattern is 121412141214121 at carrier -658
guess is: mainId 11 (0xB), subId 7 (0x7)
the pattern is 311331133113311 at carrier -658
guess is: mainId 5 (0x5), subId 7 (0x7)
the pattern is 231223122312231 at carrier -754
guess is: mainId 47 (0x2F), subId 7 (0x7)
```

Checking other multiplexes:

North Yorkshire gives correct ID's at -752 (46, 8) but wrong ID's at -656 (5, 8 and 7, 8)
Humberside gives correct ID's at -754 (47, 7) but wrong ID's at -658 (5, 7; 8, 7 and 11, 7)
SDL National gives correct ID's at -754 (47, 7) but wrong ID's at -658 (5, 7 and 11, 7)
Sheffield gives wrong ID's at -652 (5, 10 and 9, 10). The main and sub ID should be 45, 10.
D1 National gives correct ID's at -748 (45, 10) but wrong ID's at -652 (6, 10 and 9, 10)
BBC National DAB gives correct ID's at -758 (45, 5) and doesn't show anything other than that carrier.
Nottinghamshire gives correct ID's at -748 (65, 10) but wrong ID's at -652 (6, 10; 7, 10 and 10, 10).
Leeds gives correct ID's at -728 (46, 20) and doesn't show anything other than that carrier.

JvanKatwijk, 2018-01-07:
Well, in cases like this it is always helpful to create a file with raw
input and send it to me.
There is not much I can do with the data as presented.

JvanKatwijk, 2018-01-07:
As an example, in a case like the one from the picture, the algorithm
applied will definitely fail to recognize the mainId/subId
combination.
So, rather than just stating that things do not (seem to work), it is in
general more constructive to provide some data that might help
to find the cause

tghe-retford, 2018-01-07:
I've uploaded a raw input file to my Google Drive (as the zip file I put it in is approaching 300MB well beyond GitHub's supported file size limit) which contains one minute of output from the Lincolnshire multiplex which is able to display on the terminal window the issue I stated in my original post. It should be accessible to you via this link:

https://drive.google.com/file/d/10jo3swGxZLI7CqnX9zAjNulLu8czUvph/view?usp=sharing

JvanKatwijk, 2018-01-08:
> vph/view?usp=sharing

This is what I get when processing your file. It seems there are 3
transmitters with distances close to each other, so the data might be
corrupted.
Furthermore, the signal is weak, the neighbours, either noise or signal,
seem to be much stronger
j

tghe-retford, 2018-01-08:
I have recorded another raw file of Digital One (next door to the Lincolnshire multiplex), one of the strongest multiplexes I can receive as it comes from a mast 5km away line of sight. The file includes both carriers at -652 and -748.

https://drive.google.com/file/d/1r_klbma33povXBK-IP2y-bKv75j2UvrU/view?usp=sharing

JvanKatwijk, 2018-01-08:
> bKv75j2UvrU/view?usp=sharing

I see a pretty stable signal, with a start carrier detected at 748

andimik, 2018-01-08:
I have downloaded the Lincolnshire file. But the reception is very very poor. TII in Null Symbols needs a much better than you have.

JvanKatwijk, 2018-01-08:
The current approach - still experimental - does the following:
 It - obviously - computes the dft of  a part of the null symbol, a part
with the size of the useful data in the DAB blocks (i.e. Tu)
We are looking at "carriers" carriers (i.e. 1532 in mode 1), we start
scanning from carrier -768 until we might have
something, i.e. a carrier with an energy content of at least N * the
average of all carriers
Then starting from that carrier, a match is made twofold:
a. just looking at the DFT data
b. matching the DFT data with the predefined data as given in symbol 0

Since we know that the carriers in the pattern have a distance of k * 48
(however, we do not know k), but each of the 4 occurrences of the pattern
starts with a carrier with energy, there are two ways:
a. we compute for the carriers we are sure of that they exist (i.e. we look
at 4 carriers, one for each occurrence of the pattern)
b. we compute for all possible carriers in the pattern, iI.e. we look at 32
carriers, 8 for each occurrence of the pattern)

We know that the carriers in the pattern come in pairs, we know the two
elements in each pair carry the same value
and we even know the value (modulo a phase shift), so the simplest
computation is a summation over 4 elements, c*c^T

The issue is that when more than a single transmitter is received, the
signal is polluted and the matching does not result in the right result.
Too much noise may cause problems with the identification of the start
carrier.

closed 2018-02-12 by JvanKatwijk