JvanKatwijk/qt-dab #83: File paths with umlauts don't work

basicmaster, 2018-01-10:
When I want to open an `.sdr` file whose path contains german umlauts (e.g. `/tmp/mürks/test.sdr`), I get a "File not found" error.

However when I move the file to another folder without umlauts (e.g. `/tmp/murks/test.sdr`), loading works.

JvanKatwijk, 2018-01-10:
I'll look into it

andimik, 2018-01-10:
Under Windows I cannot confirm that. There it works.

The problem only is at Linux.

The error says (translated) wrong multi byte char.

basicmaster, 2018-01-10:
Correct; I forgot to mention that I tested this on Linux.

However I get no error message with more detailled information; just the following is printed to the console:

    file <filename> no legitimate sound file

andimik, 2018-01-10:
Are you sure you tested it with the latest version? Jan implemented a line showing the filepath recently.

andimik, 2018-01-10:
Update This error is only when opening rawfile, not sdr-file

basicmaster, 2018-01-10:
Yes, I compiled from the latest Git version. OS is Ubuntu MATE 16.04.

basicmaster, 2018-01-10:
Do you possibly use the precompiled version provided here at GitHub? That may make the difference.

andimik, 2018-01-10:
No, same like you. I built it.

I found the difference in rawfile.cpp line 53ff and wavfile.cpp line 55ff

JvanKatwijk, 2018-01-11:
I changed the "toLatin1()" mappings from QStrings to "toUft8()"
Testing with a wild greek filename worked, so it might be corrected now

basicmaster, 2018-01-11:
Thanks, this fixed it.

closed 2018-01-11 by basicmaster

andimik, 2018-01-11:
IQ samples as well?

JvanKatwijk, 2018-01-11:
I am afraid I do not understand the question

andimik, 2018-01-11:
Sorry, my question was not precise enough. 

I'll answer myself: Yes, also iq files with Umlauts in path now work. I thought there would be another file to change, but this was not the case.

Now there is a **new** bug. It seems there are some toUtf8() too much:

```
going to write file /tmp/qt-pictures/img/AD0B_ARBÃ_32x32.png
going to write file /tmp/qt-pictures/img/AD0B_ARBÃ_ARBÃ-Radio_ARBÃ-Radio_112.png
```

instead of 

```
going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_32x32.png
going to write file /tmp/qt-pictures/img/AD0B_ARBÖ_ARBÖ-Radio_ARBÖ-Radio_112.png
```

which was before.

Because the path to the tmp folder need not be utf8, but only latin1 and there seem to be only one ensemble (AVIS in Slovakia) using Unicode.

So, please change it again back to toLatin in `radio.cpp`, lines 690ff

JvanKatwijk, 2018-01-11:
Sorry, but I do not understand the issue.  The filenames you gave are
different, I can see that, but I do not see why it would be wrong to
convert to Utf8.
I'll have to check the behab=viour under W10, but Linux does handle Utf8
filenames

andimik, 2018-01-11:
the file should be named *ARBÖ*, not *ARBÃ*, because the Ö stands for Österreich. Ö converted to Utf8 seems to be Ã.

I've tested it under Ubuntu setting back all 3 toUtf8 (first quote) to toLatin1 (second quote). Then it worked.

andimik, 2018-01-11:
Yes, see http://www.i18nqa.com/debug/utf8-debug.html

JvanKatwijk, 2018-01-11:
OK, and what would happen if - in a different country - the file had a
decent Utf-8 encoded name?

andimik, 2018-01-11:
[image]
Rádio Litera in Slovakia seems not to have a utf8 filename ;-)

> going to write file no name

JvanKatwijk, 2018-01-12:
OK, changes. Now waiting for someone who complaints that the name of the
picture is utf8😁