JvanKatwijk/qt-dab #84: Mode 2 (L-Band Czech Republic)

andimik, 2018-01-12:
Hi Jan, 

a Dutch :-) user posted a link to a File from Czech Rep. using the L-Band and Mode 2.

http://radioforum.foren.mysnip.de/read.php?35326,1451537,1452480#msg-1452480

But qt-dab does not detect any DAB signal there

[image]

dab-cmdline also fails:

```
andreas@andiE535:~$ dab-raw-2 -M 2 -F ./LD_RTI\ cz\ DAB_2018-01-08_222026.dat
dab_cmdline V 1.0alfa,
 	                  Copyright 2017 J van Katwijk, Lazy Chair Computing
opt = M
opt = F
[...]
selecting device default
device HDA ATI HDMI: 0 (hw:0,3) seems available as 0
device HD-Audio Generic: CX20590 Analog (hw:1,0) seems available as 1
device hdmi seems available as 2
device pulse seems available as 3
device default seems available as 4
Suggested size for outputbuffer = 1280
ofdm word gestart
Period = 8000
there might be a DAB signal here

no ensemble data found, fatal
```

welle.io works, indeed (but with lots of problems) with parameter -M 2

> ./welleio.AppImage -F ./LD_RTI\ cz\ DAB_2018-01-08_222026.dat -D rawfile -M 2

But I fear the recorded file is not good enough and has a poor quality.

What do you think?

JvanKatwijk, 2018-01-13:
> 1451537,1452480#msg-1452480

This is what I get,

andimik, 2018-01-13:
And what do you get? Seems you wanted to include a screenshot, but it was not uploaded into you comment on github.com

JvanKatwijk, 2018-01-13:
sent to you via private mail

andimik, 2018-01-13:
Thanks, it did not help me. It doesn't lock.

Tried to rename it to *.iq and load it then. 

I am using version a072b33

JvanKatwijk, 2018-01-13:
No, but maybe you forgot to set the mode on 2, I do not have any problem.
It would be interesting though to get some more input dumps, since they are
mode 2 and then I can exercise an "automatic" mode finder

andimik, 2018-01-13:
Thanks, this was the solution. I FIRST have to chance to Mode 2 and then load it ...

But what about dab-raw-2? It did not detect it with parameter `-M 2`

closed 2018-01-29 by andimik

pr2git, 2018-02-10:
Hi,
Don't spent too much energy on other mode than 1 because:
Transmission modes: Only Transmission mode I is retained; the other modes are obsolete
Pr2

andimik, 2018-02-11:
No, L-Band uses mode II

lars18th, 2018-02-16:
Hi,

> Transmission modes: Only Transmission mode I is retained; the other modes are obsolete

This it's true because all other modes have **removed** from the standard.

> No, L-Band uses mode II

This it's true beucause on some territories they continue to use other modes.

In any case, I doubt new receptors will accept other modes different than Mode I. Even, if someone continues using the Mode II.