JvanKatwijk/qt-dab #85: debian package

alexmyczko, 2018-01-18:
just to let you know, i have prepared a debian package of it:

http://sid.ethz.ch/debian/qt-dab/

alexmyczko, 2018-02-03:
pressing randomly buttons crashes qt-dab with a Segmentation fault (core dumped).

Is it possible it doesn't support my two rtl-sdr usb sticks? (one is fitibit fc00012):
Bus 003 Device 011: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
Bus 003 Device 012: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

JvanKatwijk, 2018-02-04:
The supported devices are listed (somewhere) in the osmocom docs (see
osmocom.org). The 2832 based chips are supported, so if not supported, then
it must be
the tuner chip.

While it is certainly possible that there are button combinations that -
when pressed - cause the software to crash, it obviosuly should not happen.
However, it would be helpful to know what buttons are involved, I played
for a while piano on the software and it did not crash

best
jan

alexmyczko, 2018-02-04:
not sure, but is it possible to try without any working usb stick to try? because since mine are not detected, i can retry later and figure out which combinatino causes it... but is it possible it only happens if no working device is attached?

JvanKatwijk, 2018-02-04:
Great, found at least one issue: if no device is selected yet, and one
pushes the "nextChannel" button the thing crashes.
I'll fix that (Never thought of the possibility of pusjing buttons without
a devicehaving selected)

thanks
best
jan

closed 2018-02-12 by JvanKatwijk