JvanKatwijk/qt-dab #86: Audio Jitter

NomisGninnac, 2018-01-24:
From latest code, block 12B (BBC) has a very weak signal, but plays well regardless, apart from complete loss of sync every hour when the BBC switch from stereo to mono.  This is odd behaviour as it loses connection immediate after the pip sounds the BBC broadcast to mark the end of an hour!

The other problem is on block 11A, the signal is very strong, but I'm getting Audio Jitter, like a CD skipping.

I am bemused by this...

andimik, 2018-01-25:
Reconfig of the stream is not much used in Europe. Almost all muxes are static 24/7.  Could you make a rawfile or an eti-file (with dab2eti, not with Eti-stuff)?

andimik, 2018-01-25:
The other thing might be multipath reception. Could you post the Impulse Response window?

NomisGninnac, 2018-02-05:
Issue resolved as of compile of most recent commits.

Will close issue.

closed 2018-02-05 by NomisGninnac