JvanKatwijk/qt-dab #88: TII only works with 1st loaded sample file

andimik, 2018-02-10:
This problem did not appear in a27623d (3 weeks ago), but latest e57d8ba has a bug in TII guessing:

When I load a raw file and click on TII then the guessing starts. Fine so far. But: When I load another file,  the TII button does not work any more. :-(

The only workaround is quitting the program.

JvanKatwijk, 2018-02-10:
That is/was a real interesting issue. The issue was raised by someone who
noticed that whenever no device was selected, just pushing buttons
could lead to a crash of the program. The issue is that handling buttons
usually assumes the existence some classes. So, the (obvious) solution is to
disconnect the buttons (and sliders) and only connect them whenever a
device is selected and the processing chain is installed.

However, the initial startup is different from deselecting/selecting a
device. There it seemed to go wrong: some connects were made for a second
time,
leading to disfunctioning of the connection.

An attempt is made to  address this by now explicitly deconnecting devices
when deselecting/selecting a device

closed 2018-02-12 by JvanKatwijk

andimik, 2018-02-12:
ok, fixed in c6fff79