JvanKatwijk/qt-dab #89: UEP Protection level 4 and 5

andimik, 2018-02-10:
UEP Protection level 4 is shown as PL 0 and 
UEP Protection level 5 is shown as PL 1.

[image]

http://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=BEL&inputEn=6005 
(here you see, it's PL 4)

Tested with a self generated sample ETI file with ODR DabMod/DabMux:

```
Audio services
program name;country;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; prot level; code rate; language; program type

Test PL 1 AT    ;          ;A501;1;0;208;192;DAB;UEP 1;2/5;German;News;
Test PL 2 HU    ;          ;B502;2;208;168;192;DAB;UEP 2;1/2;Hungarian;Education;
Test PL 3 HR    ;          ;C503;3;376;140;192;DAB;UEP 3;3/5;Croatian;Pop Music;
Test PL 4 DE    ;          ;D504;4;516;116;192;DAB;UEP 0;7/20;German;Other Music;
Test PL 5 RO    ;          ;E505;5;632;96;192;DAB;UEP 1;2/5;Romanian;Religion;
```

JvanKatwijk, 2018-02-10:
> inputLand=BEL&inputEn=6005

Silly that no one notices that before, the protection level as such was
correct, masking of the last 2 bits before displaying the value was
obviously incorrect

pr2git, 2018-02-10:
Jan,
Be very careful the same bit masking is also applied elsewhere in the code for the protLevel.
And in there are 5 protLevel and in the case to handle it goes from 0 to 3 but protLevel is stored from 1 to 5, so I am quite sure that someprotLevel cases are not properly covered.

Pr2

pr2git, 2018-02-11:
EEP has 4 possible values but UEP has 5 possible values and RTBF is using UEP that perhaps what confuse me.
Be carefull that you need also to change the ensembleprint.cpp in various folder because it still reflect the wrong value when we save it.

closed 2018-02-12 by JvanKatwijk

andimik, 2018-02-12:
ok, fixed in 04ce449

JvanKatwijk, 2018-02-12:
I decider to clean up the mess

andimik, 2018-02-12:
Sure, no problem. 

I just added the commit where I knew it was really fixed. In other github-projects they do so in order to track what bug was fixed when ...

andimik, 2020-04-16:
Dear Jan, I fear, it has not completely fixed in 04ce449.

I found out, that in (obviously) **all** your `dab_tables.cpp` and `ensemble-printer.cpp` in all your projects there is a number missing in this line:

> const char *uep_rates  [] = {nullptr, "7/20", "2/5", "1/2", "3/5"};

(ensemble-printer.cpp in line 31)

> static const char *uep_rates [] = {"7/20", "2/5", "1/2", "3/5"};

(dab_tables.cpp in line 35)

DAB (old) has 5 (five) protection levels, not 4. PL5 is around 0,75 = 3/4.

The content output (see below from my self-made eti) still has wrong fractions!! Because UEP 3 is 1/2, but in the txt-file it's UEP 2 and UEP 5 is repeated with the wrong value for UEP 1.

See page 10 in https://tech.ebu.ch/docs/tech/tech3310.pdf

```
Audio services
program name;country;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; prot level; code rate; language; program type

Test PL 1 AT    ;          ;A501;1;0;208;192;DAB;UEP 1;2/5;German;News;
Test PL 2 HU    ;          ;B502;2;208;168;192;DAB;UEP 2;1/2;Hungarian;Education;
Test PL 3 HR    ;          ;C503;3;376;140;192;DAB;UEP 3;3/5;Croatian;Pop Music;
Test PL 4 DE    ;          ;D504;4;516;116;192;DAB;UEP 4;7/20;German;Other Music;
Test PL 5 RO    ;          ;E505;5;632;96;192;DAB;UEP 5;2/5;Romanian;Religion;
```

I wanted to patch (and fix by myself) and add 3/4 to that lines

> const char *uep_rates  [] = {nullptr, "7/20", "2/5", "1/2", "3/5", "3/4"};

resp.

> static const char *uep_rates [] = {"7/20", "2/5", "1/2", "3/5", "3/4"};

But it is still not showing the correct values.

JvanKatwijk, 2020-04-16:
Hi Andreas

Is the decoding OK?
If I read your mail correctly, the (bitRate, protection) = (192, UEP 5).
The UEP table (in the protection directory) "knows" these values.
If the decoding is correct, it is a matter of representation.
Looking at the code for the ensemble printer, I see that the mapping from
(internally) encoded UEP levels to table entries
seems right for the first 4 elements, are they right?

  else  {
                 protL = "UEP ";
                 protL. append (QString::number (h));
                 codeRate = uep_rates [h + 1];

Best
jan

andimik, 2020-04-16:
Decoding is ok, but the mapping is wrong

UEP 1 should be value 1 (= 7/20), but value 2 (= 2/5) is used
UEP 2 should be value 2 (= 2/5), but value 3 (= 1/2) is used
UEP 3 should be value 3 (= 1/2), but value 4 (= 3/5) is used
UEP 4 should be value 4 (= 3/5), but value 1 (= 7/20) is used
UEP 5 (which has to be added) should be value 5 (= 3/4), but value 2 (= 2/5) is used

So, it is +1 and breaks after the 3rd value.

JvanKatwijk, 2020-04-16:
Well, one thing in the ensemblePrinter

the value for "h" is in the range 1 .. 5, so the h+1 should be h (line 143)
  protL = "UEP ";
                 protL. append (QString::number (h));
                 codeRate = uep_rates [h];

JvanKatwijk, 2020-04-16:
second issue:

In the file dab_tables.cpp (in directory /src/support)
a. In function getCodeRate there still is the line
  return uep_rates [h & 03];     // UEP

Which is wrong, since h takes value 1 .. 5

In the same file, the uep table should be adapted
static const char *uep_rates [] = {nullptr, "7/20", "2/5", "1/2", "3/5",
"3/4"};
(since the smallest value for h is 1, a dummy zero-th element should be
added)

If these mods improve the results I'll upload the version with these mods

andimik, 2020-04-16:
I'll do some tests and will send you some PRs.

andimik, 2020-04-16:
`return uep_rates [h & 03];     // UEP`

should be 

`return uep_rates [h-1];     // UEP`

then it works :-)

```
Test PL 1 AT    ;          ;A501;1;0;208;192;DAB;UEP 1;7/20;German;News;
Test PL 2 HU    ;          ;B502;2;208;168;192;DAB;UEP 2;2/5;Hungarian;Education;
Test PL 3 HR    ;          ;C503;3;376;140;192;DAB;UEP 3;1/2;Croatian;Pop Music;
Test PL 4 DE    ;          ;D504;4;516;116;192;DAB;UEP 4;3/5;German;Other Music;
Test PL 5 RO    ;          ;E505;5;632;96;192;DAB;UEP 5;3/4;Romanian;Religion;
```

@hayguen
I've also tested it with fmlist_scan 
https://patch-diff.githubusercontent.com/raw/JvanKatwijk/qt-dab/pull/163.patch