JvanKatwijk/qt-dab #92: Blocking when we open a *sub-channel when not all data is received + negative bitrate

pr2git, 2018-02-11:
If you open up qt-dab and select a *sub-channel (data) that don't have yet received the proper DSCTy type qt-bad is blocked (everytime this happen we have this: d. DSCTy=0)

d. DSCTy = 0, d. bitRate = -26592

And we have a popup with "still insufficient data for this service"

If we wait longer and have d. DSCTy = 60 it works.

Remark also that the bitrate is displayed as a negative value.

andimik, 2018-02-14:
Jan, this is not solved.

JvanKatwijk, 2018-02-14:
I know. I think the whole idea of having subservices visible by adding a '*' to the name is not the right one and should be withdrawn, at least until I have figured out what to do with subservices

closed 2018-02-15 by JvanKatwijk