JvanKatwijk/qt-dab #93: Display MOT SlideShow (DSCTy=60) while listening to the associated radio

pr2git, 2018-02-12:
Belgian radio are sending the MOT SlideShow (SLS) on a dedicated data sub-channel, each audio have its associated data sub-channel for MOT SlideShow. Since the latest change it is now possible to select this Data sub-channel (marked with a *) and the SlideShow are properly displayed.
Now it would be nice to be able to listen to the Radio channel and get the (data) SLS displayed.

JvanKatwijk, 2018-02-15:
It would be nice - obviously - but it since it requires quite some reworking, it is not expected to happen soon

closed 2018-02-15 by JvanKatwijk

pr2git, 2018-02-17:
Why close it since it's not solved?

JvanKatwijk, 2018-02-17:
Since it will not be addressed by me

JvanKatwijk, 2018-02-17:
.... but feel free to reopen and give a pull request on the required
modifications

2018-02-17 17:37 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Since it will not be addressed by me
>
>
>> Why close it since it's not solved?
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

pr2git, 2018-02-17:
When I propose pull requests you didn't accept them so I am waiting for your own implementation.

JvanKatwijk, 2018-02-17:
Seems to me quite unfair.But as I said I am not implementing your request

pr2git, 2018-02-17:
This is not "my request" it is just an officially well documented way to send MOT which is compliant with the DAB standard but which is not supported by qt-dab.

About the "unfair" tell me what is wrong with this one then:
https://github.com/JvanKatwijk/dab-cmdline/pull/39

What is unfair is indeed implementing changes without crediting the author of it...

JvanKatwijk, 2018-02-17:
I shall explain why not:
the current approach of showing subservices is a quick hack that is pretty
aad hoc and only partially correct, I can spot a number of deficiencies.
As soon as my current project is finished I'll see if I can come with a
better - integral - solution for storing and showing suibservices.
My current project might take a month or 2 and takes virtually all time I
have .
Then there is the question of semantics: what would one (not necessarily
you) want when there are subservices involved, Should there be
a switch to select them or should they all just be made audible/visible

As soon as I have some time left and a reasonable answer on the questions,
I might consider adapting the implementation (which probably will cause
a major rewrite)

2018-02-17 17:53 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Seems to me quite unfair.But as I said I am not implementing your request
>
>
>
>> When I propose pull requests you didn't accept them so I am waiting for
>> your own implementation.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

JvanKatwijk, 2018-02-17:
> JvanKatwijk/dab-cmdline#39

Given this answer I suggest you stop using this software. Seems better for
your health and mine

pr2git, 2018-02-17:
Read EN 300 401 from chapter 6.3.5 Service component global definition up to and including: 
6.3.7.4 Packet data service components
All is explained you need to use the FIG 0/2, FIG 0/3 and FIG 0/13.
End-user don't have to select anything, the standard describe it all, when selecting the main service you can retrieve the associated sub-service that need to be used and in the data case it is the packet address (from FIG 0/3) which is needed.

JvanKatwijk, 2018-02-17:
Again, if you are not happy with the software, stop using it rather than
telling me what to do

andimik, 2018-02-27:
BTW 
London 12C also seems to use a separate subchannel 

See http://digitalbitrate.com/dtv.php?mux=9A&liste=2&live=152&lang=en