JvanKatwijk/qt-dab #95: Dynamic Label: Control code not handled (0x0A, 0x0B, 0x1F)

pr2git, 2018-02-12:
I think that the special characters on Dynamic Label are not currently properly handled.

ETSI EN 300 401 V2.1.1 - 7.4.5.2 Dynamic label

Data field: this n × 8 bit field shall contain:
• Character field: this field shall define the payload of the dynamic label segment. It shall be coded as a string of characters (maximum 16) which are chosen from the character set signalled by Charset field. If the last character of a Dynamic label segment is a multibyte character and not all bytes fit into the character field it is permitted to continue the character in the next Dynamic label segment. The first character starts with the first byte of the first dynamic label segment.

The following additional codes may be used with any of the character sets:
- Code 0A (hex) may be inserted to indicate a preferred line break.
- Code 0B (hex) may be inserted to indicate the end of a headline. Headlines shall be restricted to a
maximum length of 2 × 16 displayable characters (including hyphens introduced as a result of a control
code 1F) and may contain 0 or 1 preferred line breaks codes (Code 0A). There may not be more than
sixteen characters before any line-break and there may not be more than sixteen characters after any line-break.
- Code 1F (hex) may be inserted to indicate a preferred word break. This code may be used to display long words comprehensibly.

JvanKatwijk, 2018-02-12:
I have read the manual and decided not to handle these special characters

closed 2018-02-12 by JvanKatwijk

pr2git, 2018-02-12:
Yes you didn't handle them but you didn't suppress them from the string neither.
I think that it would be nice to suppress then those special chars from the string.