JvanKatwijk/qt-dab #100: qt-dab-alpha-1.0 freezin under windows with rtl_tcp

JGehres, 2018-04-05:
Hello Jan, 

I downloaded the latest Version of QT-DAB for Windows. Im running an SDR-Server with RTL_TCP on a Raspberry Pi. 

I can connect to the Raspberry Pi with QT-DAB but i dont get a DAB-Signal. The Spectrum locks like its dropped at the center frequency. When i change the DAB-Channel the Spectrum is freezing. Also i dont get a Sync-Signal on the DAB-Channels. After a while the hole program freezes.

While doing the same with the older Version qt-dab-0.999 erverything looks fine and i get a DAB-Signal on the same DAB-Channel as with the newer Version.

Best Regards,

Johannes Gehres

JvanKatwijk, 2018-04-06:
Thanks for the message. You are right. The "lucky" thing is that the same
behaviour is shown under Linux, so it does not depend on windows
I'll look into it and let you know

best
jan

JvanKatwijk, 2018-04-06:
It turned out to be the result of a little "optimization", which proves my
hypothesis that "optimization kills".
A new version is on github. You could give it a try

2018-04-06 11:37 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Thanks for the message. You are right. The "lucky" thing is that the same
> behaviour is shown under Linux, so it does not depend on windows
> I'll look into it and let you know
>
> best
> jan
>
>
>
>> Hello Jan,
>>
>> I downloaded the latest Version of QT-DAB for Windows. Im running an
>> SDR-Server with RTL_TCP on a Raspberry Pi.
>>
>> I can connect to the Raspberry Pi with QT-DAB but i dont get a
>> DAB-Signal. The Spectrum locks like its dropped at the center frequency.
>> When i change the DAB-Channel the Spectrum is freezing. Also i dont get a
>> Sync-Signal on the DAB-Channels. After a while the hole program freezes.
>>
>> While doing the same with the older Version qt-dab-0.999 erverything
>> looks fine and i get a DAB-Signal on the same DAB-Channel as with the newer
>> Version.
>>
>> Best Regards,
>>
>> Johannes Gehres
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980
> +31 (0) 628260355

JGehres, 2018-04-09:
I gave it a try, looks like its working now. Thanks for your support. 

ps.: nice work and keep it up

closed 2018-04-09 by JGehres